realsenseai/librealsense #4255: Can't build android library

darksharkmark, 2019-06-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | - | 
| Firmware Version                   | | 
| Operating System & Version |  Win (10) | 
| Platform                                 | Android |
| SDK Version                            | (build 2.23.0) | 
| Language                            |  unity | 
| Segment			|  VR/AR | 

### Issue Description
Based of this post: https://github.com/IntelRealSense/librealsense/issues/4155

I am looking to achieve a similar result, I do not yet have the Intel® RealSense™ Depth Camera D435i, we just ordered one and I want to get a jumpstart on things.

I got stuck at building the android library from source following this: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/readme.md

I was using the gradlew method with cmd line

this is the log I get, it looks like this file:
```
/chunked_file.cpp.o: No such file or directory
```
does not exist but I can see it in the folder? below is the full pasted log, 

any help is much appreciated 

```
C:\Users\mark\Desktop\Waxeye\Projects\QuestMR\librealsense-master\wrappers\android
λ gradlew assembleRelease

> Task :librealsense:externalNativeBuildRelease
Build realsense2 x86_64
[1/106] Building CXX object ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/chunked_file.cpp.o
[2/106] Building CXX object ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/message_instance.cpp.o
[3/106] Building CXX object ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/roscpp_serialization/src/serialization.cpp.o
[4/106] Building CXX object ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/uncompressed_stream.cpp.o
In file included from C:\Users\mark\Desktop\Waxeye\Projects\QuestMR\librealsense-master\third-party\realsense-file\rosbag\rosbag_storage\src\message_instance.cpp:28:
In file included from ../../../../../../../third-party/realsense-file/rosbag/rosbag_storage/include\rosbag/message_instance.h:150:
In file included from ../../../../../../../third-party/realsense-file/rosbag/rosbag_storage/include\rosbag/bag.h:64:
../../../../../../../third-party/realsense-file/rosbag/rosbag_storage/include\rosbag/../../../console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
  console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_DEBUG, fmt, ##__VA_ARGS__)
                                                                                         ^
../../../../../../../third-party/realsense-file/rosbag/rosbag_storage/include\rosbag/../../../console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
2 warnings generated.
[5/106] Linking CXX static library C:\Users\mark\Desktop\Waxeye\Projects\QuestMR\librealsense-master\wrappers\android\build\third-party\realsense-file\librealsense-file.a
FAILED: cmd.exe /C "cd . && C:\Users\mark\AppData\Local\Android\Sdk\cmake\3.6.4111459\bin\cmake.exe -E remove C:\Users\mark\Desktop\Waxeye\Projects\QuestMR\librealsense-master\wrappers\android\build\third-party\realsense-file\librealsense-file.a && C:\Users\mark\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\x86_64-linux-android-ar.exe qc C:\Users\mark\Desktop\Waxeye\Projects\QuestMR\librealsense-master\wrappers\android\build\third-party\realsense-file\librealsense-file.a  ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/lz4/lz4.c.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/console_bridge/src/console.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/debug.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/header.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag_player.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/buffer.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/chunked_file.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/lz4_stream.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/message_instance.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/query.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/stream.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/uncompressed_stream.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/view.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/roscpp_serialization/src/serialization.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rostime/src/duration.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rostime/src/rate.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rostime/src/time.cpp.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/roslz4/src/lz4s.c.o ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/roslz4/src/xxhash.c.o && C:\Users\mark\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\x86_64-linux-android-ranlib.exe C:\Users\mark\Desktop\Waxeye\Projects\QuestMR\librealsense-master\wrappers\android\build\third-party\realsense-file\librealsense-file.a && cd ."
C:\Users\mark\AppData\Local\Android\Sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\x86_64-linux-android-ar.exe: ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/chunked_file.cpp.o: No such file or directory
ninja: build stopped: subcommand failed.

> Task :librealsense:externalNativeBuildRelease FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':librealsense:externalNativeBuildRelease'.
> Build command failed.
  Error while executing process C:\Users\mark\AppData\Local\Android\Sdk\cmake\3.6.4111459\bin\cmake.exe with arguments {--build C:\Users\mark\Desktop\Waxeye\Projects\QuestMR\librealsense-master\wrappers\android\librealsense\.externalNativeBuild\cmake\release\x86_64 --target realsense2}
  [1/106] Building CXX object ../../../../../build/third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/chunked_file.cpp.o
[... 3 lines cut ...]
[... cut ...]
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/5.4.1/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 9s
39 actionable tasks: 2 executed, 37 up-to-date
```

ohtaisao, 2019-06-23:
@darksharkmark 
I was troubled with the same build error, but as a result of trying variously, I was able to build aar.

The following document says to select "assembleRelease" and build, but when I selected "buildDependents" I got some errors but aar was able to build.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/readme.md

I have also confirmed that the realsense pointcloud can be seen on OculusQuest with the aar that I built.

unagiHuman, 2019-06-23:
@darksharkmark
I was troubled with the same build error, but as a result of trying variously, I was able to build aar.

The following document says to select "assembleRelease" and build, but when I selected "buildDependents" I got some errors but aar was able to build.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/readme.md

I have also confirmed that the realsense pointcloud can be seen on OculusQuest with the aar that I built.

closed 2019-07-10 by RealSenseCustomerSupport

shivasku82, 2020-03-09:
for me issue is because of max path length , keep project folder path small