realsenseai/librealsense #4257: Deepcopy image data when pulling frames from the camera

iggy12345, 2019-06-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System | windows |
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { 2.21 }                          | Language | { python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I struggled for a long time to try to record the data from my D435 camera into a separate video file, and I couldn't figure out why the program kept freezing after 30 frames or so, until I realized it's because when I pull frames off of the camera, it's still being held in some sort of ring buffer on the camera, and deepcopying the data before I stores it into an array solved the problem, so... Feature request: please deepcopy the frame data from the camera so that I dont run into memory issues like this again in the future.

iggy12345, 2019-06-21:
Sorry if this isnt well thought out, I typed it in 5 minutes on my phone

dorodnic, 2019-06-22:
Hi @iggy12345 
That's fine, what you are describing is clear. 
However, if I understand correctly, the "crash" happens because `wait_for_frames` refused to provide any additional frames. This behavior was discussed on this forum, and the bottom line is that it's by design. If you do want to keep SDK frames for longer, you can call `frame.keep()` method. By default, the SDK limits the number of frames that can be used at any given time to prevent unbound latency due to slow processing.

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@iggy12345 Any other questions? Looking forward to your update. Thanks!

iggy12345, 2019-07-12:
That did solvsolve my issue, sorry, forgot to close it

closed 2019-07-12 by iggy12345