realsenseai/librealsense #4258: Real-time 3D reconstruction using T265

mola1129, 2019-06-21:
Hi,
We can compute stereo depth from T265's two cameras like [this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py).
Is it possible to reconstruct the 3D room with textured meshes using a T265 device + tablet PC in real-time?  It doesn't need to be high accuracy for me and  I want to display T265's pose too similar to this [video](https://www.youtube.com/watch?v=JotzEDwRA8M).

Thanks.

MartyG-RealSense, 2019-06-21:
Intel did something similar to that Project Tango video in a 2015 demo with a prototype RealSense Smartphone Development Kit (a cancelled Android smartphone with a RealSense ZR300 stereo camera and Project Tango).

https://www.youtube.com/watch?v=plwYJqz04GE

https://www.youtube.com/watch?v=Vpa8vKsapFk

Project Tango eventually got cancelled by Google and replaced by ARCore.

https://developers.google.com/ar/

Since you have computed depth from the T265, and the T265 has an IMU component like the RealSense D435i depth camera model, I am wondering if you might be able to make use of the guide below for generating SLAM with a D435i and ROS and apply it to your T265.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i?language=en_US

mola1129, 2019-06-24:
Thanks. I will try it.

closed 2019-06-24 by mola1129