realsenseai/librealsense #4259: Access denied in Windows 10 Pro 1903 (build 18362.175)

ksze, 2019-06-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | *should* be 5.11.6.250, as Windows offered to upgrade the firmware when I plugged it in | 
| Operating System & Version |   Windows 10 Pro, 1903, build 18362.175  | 
| Platform                                 | PC |
| SDK Version                            |  2.21.0                          | 
| Language                            |  not applicable                          | 
| Segment			|  others                          | 

### Issue Description
Intel.RealSense.Viewer.exe can't open the sensor, access denied (0x80070005)

I have already enabled camera and location access as per #3644 / #3381 / #3879 .

Here's the log:

```
 21/06 17:26:17,097 WARNING [10528] (ds5-factory.cpp:678) DS5 group_devices is empty.
 21/06 17:26:17,097 WARNING [10528] (ds5-factory.cpp:678) DS5 group_devices is empty.
 21/06 17:26:21,235 WARNING [10528] (ds5-factory.cpp:678) DS5 group_devices is empty.
 21/06 17:26:21,504 WARNING [10528] (ds5-factory.cpp:678) DS5 group_devices is empty.
 21/06 17:26:21,575 ERROR [18912] (types.h:243) MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070005: "Access is denied."
 21/06 17:26:21,575 WARNING [18912] (rs.cpp:279) null pointer passed for argument "device"
 21/06 17:27:09,713 ERROR [3900] (types.h:243) MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070005: "Access is denied."
 21/06 17:27:09,713 WARNING [3900] (rs.cpp:279) null pointer passed for argument "device"
 21/06 17:42:44,164 ERROR [17360] (types.h:243) MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070005: "Access is denied."
 21/06 17:42:44,169 WARNING [17360] (rs.cpp:279) null pointer passed for argument "device"
```

ev-mp, 2019-07-01:
@ksze hello,
Please check that camera hardware is enabled in Windows'  `Camera Privacy Settings` 
[image] #3381, #2572

ksze, 2019-07-02:
HI @ev-mp, like I wrote above, I have already followed the advice in other tickets (#3644 / #3381 / #3879) to enable location and camera in Windows' privacy settings, for both Microsoft Store apps and "traditional" desktop apps, to no avail.

ev-mp, 2019-08-05:
@ksze, for D435i you also need to check another Win10 property  - `location services` are correctly configured to allow for IMU sensor ([link](https://github.com/IntelRealSense/librealsense/issues/3644#issuecomment-479043117))

ksze, 2019-08-06:
@ev-mp I had also already enabled location access (see original issue description).

I’m going to bold that part in case anybody else overlooks it, heh.

PaulBremner, 2019-08-06:
Hoping someone has found a solution to this. I get the same issue, and like the OP I have camera and location privacy settings enabled (though as I am using a 415 the location shouldn't matter).

I have updated to the latest drivers and Realsense viewer 2.25, and done a full restart.

I am able to access the RGB camera through the windows camera app so it is clearly connected and working. There must be some other permission setting somewhere, but I can't find any.

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

Hi @ksze,

Please provide us with the following details about your system configuration :

how many cameras are in active use in the system (you have mentioned that under "Sensors" in Device Manager you can see 5 of "HID Sensor Collection V2")? Please note, that for one D435i camera you should see only one instance of "HID Sensor Collection V2" in Device Manager.

for each camera please provide :
  camera model (D435i,etc)
  FW version

how the camera(s) is/are connected (direct connection, USB hub, etc.)

RealSense Viewer version

Thank you!

ksze, 2019-11-16:
@RealSenseCustomerSupport Unfortunately, I no longer work at the company where the D435i is used, so I don't have access to a D435i and somebody else will need to pick up the slack. Perhaps @PaulBremner can provide the info?

RealSenseCustomerSupport, 2019-11-16:
----------------------------------------------

@PaulBremner

Can you please provide us with the details about your system configuration?

Thank you!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@PaulBremner

Do you still need help with this question?

closed 2019-12-20 by RealSenseCustomerSupport