realsenseai/librealsense #4260: Not able to play rosbag with ros command

ShettyHarapanahalli, 2019-06-21:
| | |
|---|---|
|**librealsense**|2.22.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|745412070314|
|**Firmware Version**|05.10.03.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Hi,

Want to record data from real sense and generate a video file for further processing. I am able to save the recording in the bag format and play it back with the realsense viewer, but when I try to play it with the command rosbag play <file name> I get the below error - 
[ INFO] [1561133313.635821050]: Opening 20190621_170754.bag

[FATAL] [1561133313.653045811]: Character [ ] at element [32] is not valid in Graph Resource Name [/device_0/sensor_0/option/Enable Auto Exposure/value].  Valid characters are a-z, A-Z, 0-9, / and _.

How can I reslove this, I want to extract video from the RGB camera (at the cameras resloution).

dorodnic, 2019-06-22:
[realsense-ros](https://github.com/IntelRealSense/realsense-ros) node has a parameter that would play SDK .bag file.

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@ShettyHarapanahalli Did you get it through? Looking forward to your update. Thanks!

quattrinili, 2019-07-16:
The problem is related to the fact that, as shown in the error, some of the topics include the space character, which is not allowed in a topic name. I did not check the code in librealsense, however, a quick workaround using ROS is just to use the `filter` tool in `rosbag` to get only the images (here more information http://wiki.ros.org/rosbag/Commandline#rosbag_filter). For example, to get the IMU data and the image data (properly replacing the elements in <>):

```
rosbag filter <input.bag> <output.bag> "topic == '/device_<device_id>/sensor_<sensor_id>/<stream_type>_<stream_id>/imu/data' or topic == '/device_<device_id>/sensor_<sensor_id>/<stream_type>_<stream_id>/image/data'"
```

RealSenseCustomerSupport, 2019-07-17:
----------------------------------------------

@quattrinili Thanks for the good point!  @ShettyHarapanahalli  Could you please have a try with quattrinili's workaround? Thanks!

ShettyHarapanahalli, 2019-07-17:
I used the python wrapper, the example below helped me do the job for me - 
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py 

@quattrinili thanks for the command.

closed 2019-07-17 by ShettyHarapanahalli

ShettyHarapanahalli, 2019-07-30:
Just sharing for reference, the below tool helps extract data from ROS bag file - 
https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

Udaykiran87, 2021-06-21:
> The problem is related to the fact that, as shown in the error, some of the topics include the space character, which is not allowed in a topic name. I did not check the code in librealsense, [...]

I am using ros1 bridge to read the rosbag file. Unfortunately i am unable to play the rosbag file using ros2 bag play command.
It gave error like bellow:

```
what():  Invalid topic name: topic name token must not start with a number:
  '/device_0/sensor_1/Color_0/tf/0'
```

Then i tried with filter option, but i got below error message. II must play the rosbag files to be able to publish image topics.

```
rosbag filter ./Inputs/realsense_video/20210615_134517.bag ./Inputs/realsense_video/20210615_134517=out.bag "topic == '/device_0/sensor_1/Color_0/image/data' or topic == '/device_0/sensor_0/Depth_0/image/data'"

Traceback (most recent call last):
  File "/opt/ros/noetic/bin/rosbag", line 34, in <module>
    import rosbag
  File "/opt/ros/noetic/lib/python3/dist-packages/rosbag/__init__.py", line 33, in <module>
    from .bag import Bag, Compression, ROSBagException, ROSBagFormatException, ROSBagUnindexedException
  File "/opt/ros/noetic/lib/python3/dist-packages/rosbag/bag.py", line 69, in <module>
    import rospy
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 54, in <module>
    import rospy.core
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/core.py", line 75, in <module>
    from rosgraph_msgs.msg import Log
ImportError: cannot import name 'Log' from 'rosgraph_msgs.msg' (/opt/ros/foxy/lib/python3.8/site-packages/rosgraph_msgs/msg/__init__.py)
```

Can any one suggest what to do?