realsenseai/librealsense #4263: T265 persitent AR: save/load_relocalization_map

ar-ml, 2019-06-22:
| Camera Model                       | T265 | 
| Operating System & Version |   Win 10 for testing / Android for production | 
| Language                            |  Unity/C#  |                           
| Segment			|  Smartphone/AR | 

### Usecase
Using T265 for persistent AR experiences in a Museum on Android tablets.

### Questions
1. Is it possbile by now to save/load relocalization maps to achieve persistent AR experiences?
(I want to save area maps and relocalise after completely restarting the device)
2. If yes: How accurate and reliable is the relocalisation after device restart? How long would relocalisation aprox. take in a 100 square meter room?
3. Is this feature part of your Unity Wrapper? Is there an example for saving / loading maps in the wrapper? Where can I find more info / a documentation for this feature?
4. As I understand Andoid support is planed but not available yet. Could you maybe give me a little hint... should we bet on T265 if we have to deploy on Android by the end of this year?

Thank you very much!

JimmyFroost, 2019-06-26:
Hi,

I am looking forward to answers and solutions regarding this issue as well, especially the save/load relocalization map to avoid bad trajectory.
I use Matlab wrapper, I hope it will also get some useful info and examples.

Thanks.

RealSenseCustomerSupport, 2019-12-24:
----------------------------------------------

Hi @ar-ml

There has been some updates to relocalization since this thread started.  Here is a link to the latest release notes available for librealsense 2.31.0 https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2310

closed 2020-09-11 by RealSenseSupport