realsenseai/librealsense #4266: D435 development kit

buctfy, 2019-06-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.0                           | 
| Language                            |  C++/opencv                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have got object coordinate by using realsense D435 in laptop（opencv ,vs15,realsense sdk 2),but I want to realize this function in development kit(because laptop is big ).I know Intel RealSense R200  have relative development kit（https://www.mouser.com/new/Intel/intel-realsense-robotic-dev-kit/）。Is there recommended development kit with D435?

MartyG-RealSense, 2019-06-23:
I believe you are asking about a recommendation for a small development board like the Up Board in the robotics kit?

The 400 Series cameras can work with any Intel or Arm processor, though the camera's capabilities will be very limited on boards that only have a USB 2 port instead of USB 3.

There are plenty of boards to choose from because of the 400 Series' flexibility.  A good choice for the full USB 3 experience is the Up Squared, a newer and more powerful version of the original Up Board in the robotics kit.  It has been confirmed to work with up to two 400 Series cameras attached to it.

https://up-board.org/upsquared/specifications/

buctfy, 2019-06-24:
> I believe you are asking about a recommendation for a small development board like the Up Board in the robotics kit?
>
> The 400 Series cameras can work with any Intel or Arm processor, though the [...]

I used face recognition dnn model,so code ran a little slowly in my laptop ,I want to know weather up-board  can detect face in real time

MartyG-RealSense, 2019-06-24:
Comparing the spec of the Up Squared to the spec of an average budget business laptop, it looks as though Up Squared would provide comparable performance to a budget laptop.

https://up-board.org/upsquared/specifications/

I notice that Up are now selling a modern equivalent to the old robotics kit that pairs an Up Squared with a D435i camera.

https://up-shop.org/home/307-up-squared-robomaker-developer-kit.html

buctfy, 2019-06-26:
> Comparing the spec of the Up Squared to the spec of an average budget business laptop, it looks as though Up Squared would provide comparable performance to a budget laptop.
>
> I notice that Up are [...]

Up Squared can install win 10,and I have got the object coordinates with opencv ,realsense D435 ,vs15 in windows 10.Now I plan to remove the code to Up Squared ,and hope realsense D435  to work without laptop.(Turn on the  Up Squared 's switch,realsense get object coordinates automatically).I don't know how to do.Can you give me some advise？

MartyG-RealSense, 2019-06-26:
I would think that the most straightforward way to do it would be to set Windows 10 on the Up Squared to automatically start your program on boot-up so that when it completes power-up, it launches your program and captures the coordinates with the camera.

Information about adding an application to Windows 10's start-up launcher can be found in the link below.

https://support.microsoft.com/en-gb/help/4026268/windows-10-change-startup-apps

buctfy, 2019-06-26:
> I would think that the most straightforward way to do it would be to set Windows 10 on the Up Squared to automatically start your program on boot-up so that when it completes power-up, it launches [...]

you mean I need to  install win10,opencv and vs15 in upboard  and set  exe automatically start.I can get object coordinates without laptop.

MartyG-RealSense, 2019-06-26:
If you install windows 10 on an Up Board then it should boot like a Windows desktop PC, and allow you to set your program to start automatically.

You should also install on the Up Board any software that your application relies upon to work, such as OpenCV.

buctfy, 2019-06-28:
> If you install Windows 10 on an Up Board then it should boot like a Windows desktop PC, and allow you to set your program to start automatically.
> 
> You should also install on the Up Board any software that your application relies upon to work, such as OpenCV.

Thanks a lot

closed 2019-06-28 by buctfy