realsenseai/librealsense #4270: Can not install on Nvidia Jetson TX2

ArtlyStyles, 2019-06-24:
I tried to install librealsense2 on Nvidia Jetson TX2 following instruction here 
    https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
However, when I run 
    sudo apt-get install librealsense2-dkms  
I got the following error:
   E: Unable to locate package librealsense2-dkms
How can I fix the problem?

Thanks,

yx0123, 2019-06-26:
You have to do manual installation for the TX2 following the instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

closed 2019-07-12 by RealSenseCustomerSupport

ArtlyStyles, 2019-07-15:
Are you sure follow ubuntu instructions, not the jetson instruction?

yx0123, 2019-07-15:
> Are you sure follow ubuntu instructions, not the jetson instruction?

Yes, it worked for me.