realsenseai/librealsense #4271: Successfully installed librealsense on Arch (modified files to work with kernel 5.1.12)

UsaidPro, 2019-06-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |  Arch Linux  | 
| Kernel Version (Linux Only)    | 5.1.12 | 
| Platform                                 | PC (Dell XPS 15 9560) |
| SDK Version                            | 2.23.0 | 
| Language                            | N/A | 
| Segment			| others | 

### Issue Description
I managed to successfully install librealsense on the latest Arch Linux (as of 06/23/2019) and wanted to share my installation method with the Realsense community.
I verified that my installation worked using `realsense-viewer`:
[image]

I initially followed the steps outlined in [Installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). I installed via pacman `openssl`, `libusb`, 'mesa', and `gtk3` (I am not sure if I truly needed to install those packages, but installed them as they were listed as dependencies). I ran `./scripts/setup_udev_rules.sh` as instructed.

My issues started at building and patching the kernel. I installed `base-devel` and `linux-headers` and ran `cd ./scripts/`. However, when running `./patch-arch.sh` I ran into the issue that apparently the 3 urls provided [here](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-arch.sh#L25) were intended for kernel versions v4.x. I needed to update the urls to v5.x and had to drop the third url. I also needed to manually edit the `realsense-camera-formats.patch` file to adjust to kernel 5.1.12.

Finally, there was an error I encountered at the [`sudo modprobe uvcvideo`](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-arch.sh#L112): `modprobe: ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)`. Interestingly, `dmesg` reported no errors. Nevertheless, I solved it by moving the `uvcvideo.ko.xz` file created in the scripts/ folder to `/usr/lib/modules/5.1.12-arch1-1-ARCH/kernel/drivers/media/usb/uvc/`. I believe that this is an error due to [`sudo cp uvcvideo.ko.xz $MODULE_NAME`](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-arch.sh#L112) which should be `sudo cp uvcvideo.ko.xz $MODULE_NAME.xz`, but didn't modify my gist since I had a working solution.

After this, I finally finished the rest of the steps (make build directory, cmake, make, make install, etc) and have a working Intel Realsense SDK!
[Link to my gist with modified patch-arch.sh and realsense-camera-formats.patch](https://gist.github.com/UsaidPro/3b9020d6dcf14f4ac2202ff5e5332385). I can create a pull request if needed.

UsaidPro, 2019-06-24:
I missed another error I had at the end: `modprobe: ERROR: could not insert 'uvcvideo': Exec format error`. Looking at `dmesg`, this was due to `uvcvideo: version magic '5.1.12 SMP preempt mod_unload ' should be '5.1.12-arch1-1-ARCH SMP preempt mod_unload '`. This was easily solved by doing `sudo modprobe --force uvcvideo` and `rm -rf kernel/` for cleanup.

dorodnic, 2019-06-24:
Hi @UsaidPro 
Thank you for sharing, let us know if you encounter any problems

closed 2019-07-11 by RealSenseCustomerSupport