realsenseai/librealsense #4272: Announcement: Raspberry Pi 4 Model B with two USB 3 ports now available

MartyG-RealSense, 2019-06-24:
Hi everyone,

A new 'Raspberry Pi 4 Model B' development board is now available to purchase that has two USB 3 ports.

https://www.raspberrypi.org/products/raspberry-pi-4-model-b/

nikhilkoditala, 2019-06-29:
Hi @MartyG-RealSense 
Thanks for sharing this info. I want to connect and sync around 8 D415 cameras. To transfer frames, I am thinking to use one raspberry pi 4 model to connect to 2 cameras via usb 3 and a ethernet hub to connect raspberry pi to central system. So, I will be having 4 raspberry pi's which connects 8 cameras to central system. Apart from this I will be using a sync cable to sync all the cameras. Is it possible to do so? or can you suggest any better approach.

MartyG-RealSense, 2019-06-29:
Your proposed setup sounds like the EtherSense project that was created by an Intel RealSense employee, with two cameras on each board instead of one.

https://github.com/krejov100/EtherSense

If the spec of the board is equal to or greater than the original-model Up Boards used in the EtherSense example (Pi 4 is superior) then it should be able to cope with 2 cameras on the same board.  Two D415 cameras have been successfully used on the 'Up Squared' board (a newer and more powerful Up Board model).  

An Intel Support team member thought that even an original Up Board should be able to cope with two cameras so long as a mains-powered USB 3 hub was used with the board in order to stably meet the power needs of the cameras.  The 400 Series cameras have a 'Vision Processor D4' hardware board inside them to handle processing of captured camera data and reduce the need for a powerful CPU / GPU on the computing device that the cameras are attached to.  So a low-spec board should be able to handle two cameras as long as the resolutions being used are not excessively high.

Given that the Pi 4 has performance equal to an entry-level PC though, in theory you should be able to get good camera performance out of it.

SirDifferential, 2019-07-03:
I'll write in this issue. I'll create a new one if you wish.

I'm currently trying to get a D435 working on an rpi4 with the latest raspbian-lite image. As far as I know, a new kernel must be compiled since the required drivers aren't enabled in the stock kernel. I also had to port the patches to the latest raspberry kernel, and I'm not 100% if I did everything correctly. A few changes from the ubuntu patches had already been committed to upstream. Here's the modified patch that can be applied on the latest raspberry kernel: https://raw.githubusercontent.com/Delicode/librealsense/master/scripts/realsense_raspberry4.patch

To begin with, here's the steps I took so far:

Backup everything in `/boot` somewhere else so you can restore your system if the new kernel doesn't boot. Then install required packages and get the kernel sources

```
sudo apt-get install build-essential bison flex bc git libncurses-dev libssl-dev libprotobuf-dev cmake libusb-1.0-0-dev
cd ~
git clone -b rpi-4.19.y https://github.com/raspberrypi/linux.git --depth 1 rpi-4.19.y
cd rpi-4.19.y
```

Apply the kernel patches:

```
wget https://raw.githubusercontent.com/Delicode/librealsense/master/scripts/realsense_raspberry4.patch
patch -p1 < realsense_raspberry4.patch
```

Customize the config to enable the drivers which librealsense requires:

```
make bcm2711_defconfig
make menuconfig
```

Enable the following drivers if they aren't enabled:

* HID_SENSOR_HUB in Device Drivers -> HID support -> HID bus support -> Special HID drivers
* HID_SENSOR_ACCEL_3D in Device Drivers -> Industrial I/O support -> Accelerometers
* HID_SENSOR_GYRO_3D in Device Drivers -> Industrial I/O support -> Digital gyroscope sensors
* USB_VIDEO_CLASS in Device Drivers -> Multimedia support -> Media USB Adapters
* VIDEO_V4L2 in Device Drivers -> Multimedia support

Compile the kernel and install it:

```
make ARCH=arm -j4 zImage modules dtbs
sudo make ARCH=arm modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/kernel7l.img
sudo reboot
```

If the raspberry doesn't boot, restore the backupped `/boot` folder.

While following `/var/log/kern.log`, enable the new drivers:

```
sudo modprobe videodev
sudo modprobe uvcvideo
sudo modprobe hid_sensor_accel_3d
sudo modprobe hid_sensor_gyro_3d
```

If no errors appear, add the drivers in `/etc/modules` as follows:

```
videodev
uvcvideo
hid_sensor_accel_3d
hid_sensor_gyro_3d
``` 

Compile librealsense. I'm not building the examples due to large dependencies. They also don't currently link against librealsense.so. To fix this, add `-latomic` in the linker flags.
[... 110 lines cut ...]
Jul  3 10:19:20 raspberrypi kernel: [   48.922849] [<c076995c>] (generic_probe) from [<c0760118>] (usb_probe_device+0x34/0x4c)
Jul  3 10:19:20 raspberrypi kernel: [   48.925108] [<c0760118>] (usb_probe_device) from [<c06d62d4>] (really_probe+0x20c/0x2c8)
Jul  3 10:19:20 raspberrypi kernel: [   48.927378] [<c06d62d4>] (really_probe) from [<c06d6564>] (driver_probe_device+0x70/0x184)
Jul  3 10:19:20 raspberrypi kernel: [   48.929637] [<c06d6564>] (driver_probe_device) from [<c06d6828>] (__device_attach_driver+0xc0/0xe4)
Jul  3 10:19:20 raspberrypi kernel: [   48.931909] [<c06d6828>] (__device_attach_driver) from [<c06d4300>] (bus_for_each_drv+0x90/0xd4)
Jul  3 10:19:20 raspberrypi kernel: [   48.934176] [<c06d4300>] (bus_for_each_drv) from [<c06d6040>] (__device_attach+0xe0/0x148)
Jul  3 10:19:20 raspberrypi kernel: [   48.936450] [<c06d6040>] (__device_attach) from [<c06d68a8>] (device_initial_probe+0x1c/0x20)
Jul  3 10:19:20 raspberrypi kernel: [   48.938743] [<c06d68a8>] (device_initial_probe) from [<c06d5324>] (bus_probe_device+0x94/0x9c)
Jul  3 10:19:20 raspberrypi kernel: [   48.941034] [<c06d5324>] (bus_probe_device) from [<c06d1be8>] (device_add+0x37c/0x624)
Jul  3 10:19:20 raspberrypi kernel: [   48.943318] [<c06d1be8>] (device_add) from [<c075425c>] (usb_new_device+0x284/0x44c)
Jul  3 10:19:20 raspberrypi kernel: [   48.945596] [<c075425c>] (usb_new_device) from [<c0755d88>] (hub_event+0xe04/0x11a8)
Jul  3 10:19:20 raspberrypi kernel: [   48.947866] [<c0755d88>] (hub_event) from [<c023d470>] (process_one_work+0x23c/0x518)
Jul  3 10:19:20 raspberrypi kernel: [   48.950166] [<c023d470>] (process_one_work) from [<c023e534>] (worker_thread+0x60/0x5b8)
Jul  3 10:19:20 raspberrypi kernel: [   48.952460] [<c023e534>] (worker_thread) from [<c0243f70>] (kthread+0x16c/0x174)
Jul  3 10:19:20 raspberrypi kernel: [   48.954755] [<c0243f70>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
Jul  3 10:19:20 raspberrypi kernel: [   48.957046] Exception stack(0xefa71fb0 to 0xefa71ff8)
Jul  3 10:19:20 raspberrypi kernel: [   48.959317] 1fa0:                                     00000000 00000000 00000000 00000000
Jul  3 10:19:20 raspberrypi kernel: [   48.961610] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jul  3 10:19:20 raspberrypi kernel: [   48.963891] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jul  3 10:19:20 raspberrypi kernel: [   48.966142] Code: e50b0038 e50b3054 e2433078 e50b3050 (15943000) 
Jul  3 10:19:20 raspberrypi kernel: [   48.968481] ---[ end trace 80f1a4e587dbfe6f ]---
```

Attached is a full kern.log for one boot, just in case there's something at boot time you need to see.
[kern.log](https://github.com/IntelRealSense/librealsense/files/3354250/kern.log)

nikhilkoditala, 2019-07-03:
@SirDifferential I don't know about your error, probably you can go through the below link which may be helpful.
https://dev.intelrealsense.com/docs/using-depth-camera-with-raspberry-pi-3

SirDifferential, 2019-07-05:
I think I solved part of the problem. These commits:

* https://github.com/raspberrypi/linux/commit/3bc85817d7982ed53fbc9b150b0205beff68ca5c
* https://github.com/raspberrypi/linux/commit/088ead25524583e2200aa99111bea2f66a86545a

Change how UVC drivers add their quirks into the driver description by using the UVC_QUIRK_INFO macro and then change the struct size. I added that macro to the Realsense devices and the driver no longer crashes. Here are the changes: https://github.com/Delicode/librealsense/commit/cbe84c7400fa9450e1db4d616494957f35c91d70

However, the sensor doesn't work. Here's what kern.log has upon plugging in the device:

```
Jul  5 08:27:30 raspberrypi kernel: [   51.711813] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
Jul  5 08:27:30 raspberrypi kernel: [   51.742621] usb 2-1: New USB device found, idVendor=8086, idProduct=0b07, bcdDevice=50.b6
Jul  5 08:27:30 raspberrypi kernel: [   51.742637] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul  5 08:27:30 raspberrypi kernel: [   51.742649] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435
Jul  5 08:27:30 raspberrypi kernel: [   51.742662] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435
Jul  5 08:27:30 raspberrypi kernel: [   51.742673] usb 2-1: SerialNumber: 824613021757
Jul  5 08:27:30 raspberrypi kernel: [   51.755188] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
Jul  5 08:27:30 raspberrypi kernel: [   51.763890] uvcvideo: Unable to create debugfs 2-2 directory.
Jul  5 08:27:30 raspberrypi kernel: [   51.764295] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
Jul  5 08:27:30 raspberrypi kernel: [   51.764311] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
Jul  5 08:27:30 raspberrypi kernel: [   51.764326] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
Jul  5 08:27:30 raspberrypi kernel: [   51.764685] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/input/in
put2
Jul  5 08:27:30 raspberrypi kernel: [   51.766025] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
Jul  5 08:27:30 raspberrypi kernel: [   51.769303] uvcvideo: Unable to create debugfs 2-2 directory.
Jul  5 08:27:30 raspberrypi kernel: [   51.769676] uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
Jul  5 08:27:30 raspberrypi kernel: [   51.769691] uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
Jul  5 08:27:30 raspberrypi kernel: [   51.769706] uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!
```

Here's what my program says upon trying to access the device:

```
Depth stream created: 1280 x 720 at 30 FPS
Color stream created: 1920 x 1080 at 30 FPS
streams created
Device serial is: 825312070308
camera product id: 0B07
Changing preset to High Accuracy
streams started
rs_error was raised when calling rs2_get_video_stream_intrinsics(from:0xa9409e88, intr:0xb2f09d90):
get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
Failed getting depth stream intrinsics
```

And kernel says the following at the same time:

```
Jul  5 08:27:59 raspberrypi kernel: [   81.132582] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Jul  5 08:28:01 raspberrypi kernel: [   82.337162] usb 2-1: USB disconnect, device number 2
Jul  5 08:28:01 raspberrypi kernel: [   82.807323] usb 2-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
Jul  5 08:28:01 raspberrypi kernel: [   82.838108] usb 2-1: New USB device found, idVendor=8086, idProduct=0b07, bcdDevice=50.b6
Jul  5 08:28:01 raspberrypi kernel: [   82.838124] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul  5 08:28:01 raspberrypi kernel: [   82.838137] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435
Jul  5 08:28:01 raspberrypi kernel: [   82.838150] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435
Jul  5 08:28:01 raspberrypi kernel: [   82.838161] usb 2-1: SerialNumber: 824613021757
Jul  5 08:28:01 raspberrypi kernel: [   82.842038] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
Jul  5 08:28:01 raspberrypi kernel: [   82.846255] usb 2-1: WARN: Max Exit Latency too large
Jul  5 08:28:01 raspberrypi kernel: [   82.846272] usb 2-1: Could not enable U2 link state, xHCI error -22.
Jul  5 08:28:01 raspberrypi kernel: [   82.846951] uvcvideo: Unable to create debugfs 2-3 directory.
Jul  5 08:28:01 raspberrypi kernel: [   82.848169] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
Jul  5 08:28:01 raspberrypi kernel: [   82.848188] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
Jul  5 08:28:01 raspberrypi kernel: [   82.848203] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
Jul  5 08:28:01 raspberrypi kernel: [   82.848982] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/input/in
put3
Jul  5 08:28:01 raspberrypi kernel: [   82.851382] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
Jul  5 08:28:01 raspberrypi kernel: [   82.854756] uvcvideo: Unable to create debugfs 2-3 directory.
[... cut ...]
Jul  5 08:28:11 raspberrypi kernel: [   93.185432] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -110 (exp. 1024).
Jul  5 08:28:12 raspberrypi kernel: [   93.795367] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -110 (exp. 1024).
Jul  5 08:28:13 raspberrypi kernel: [   94.405296] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -110 (exp. 1024).
Jul  5 08:28:13 raspberrypi kernel: [   95.015268] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -110 (exp. 1024).
```

Gagnon06, 2019-07-15:
@SirDifferential Were you able to make it work on the Raspberry Pi 4? I have similar issues as you (https://github.com/IntelRealSense/librealsense/issues/4375#issuecomment-509984739).

SirDifferential, 2019-07-16:
@Gagnon06 No, I was not. I'm wondering if this is a kernel problem in 4.19. I updated my x86 laptop to Debian Buster and compiled 4.19.59 with the librealsense patches applied, but I'm getting pretty much identical results to running the code on raspberry 4: The basic, minimal examples sometimes work, increasing resolution to 1280x720 causes more problems, and the more advanced commands such as presets, emitters and getting fov cause the device to disconnect.

I'm afraid this needs a look from someone who knows something about the USB and UVC drivers.

Gagnon06, 2019-07-16:
@SirDifferential I may I found a way to patch the kernel. Intel has recipes for Yocto, which the last release supports kernel 4.19. I applied the patches from here : https://github.com/IntelRealSense/meta-intel-realsense/tree/master/recipes-kernel/linux/linux-yocto and recompiled the modules (the same way you described in this comment : https://github.com/IntelRealSense/librealsense/issues/4272#issuecomment-508015848). The rs-depth example seems to be more reliable now, but I still have some problems with my own software.

Error: `hwmon command 0x10 failed`

I will continue to investigate.

SirDifferential, 2019-07-17:
I applied the yocto patches to debian buster (https://github.com/Delicode/librealsense/blob/master/scripts/debianbuster-4.19.59.patch) and raspbian (https://github.com/Delicode/librealsense/blob/master/scripts/realsense_raspberry4.patch) kernels, and the results are largely the same. It might be that the 1280x720 depth feed works better, although sometimes on ^C the the kernel prints this when I ^C the rs-depth example:

```
Jul 17 08:19:07 raspberrypi kernel: [  895.260494] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260528] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260568] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260574] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260581] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260609] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260614] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260649] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260654] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260658] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260688] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260728] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260733] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260769] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260812] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260849] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260889] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260894] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260929] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.260969] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261009] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261014] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261049] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261089] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261129] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261134] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261169] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261174] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261209] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261214] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261249] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261254] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261290] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261329] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261370] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261375] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261379] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261410] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261450] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
Jul 17 08:19:07 raspberrypi kernel: [  895.261490] xhci_hcd 0000:01:00.0: WARN Event TRB for slot 2 ep 0 with no TDs queued?
[... cut ...]
I also managed to get the device running on x86 Debian Buster in the realsense viewer with these patches, although it's spotty at best. I get errors like this when I try to control assorted knobs in the UI:

```
get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
```

But sometimes the D435 still reacts to the instructions even if an error is shown. Unfortunately some controls don't seem to work, namely the presets. The D435 is detected as a USB 3.2 device though, so that eliminates a few possibilities.

Gagnon06, 2019-07-17:
Just tested the rs-depth example with a resolution of 1280x720 and it seems to work fine. As for the problem I had before (`hwmon command 0x10 failed`), it was when we were trying to get the info (`rs2::device::get_info`) on the device too soon after setting it up and enabling the feed. Now the camera is working well, the only problem I see is a kernel log that shows up : 
```
[  198.567278] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  199.261753] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  199.312128] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  199.462983] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  199.513449] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  200.065930] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  200.116269] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

SirDifferential, 2019-07-18:
That's a good find. I switched over the development branch with that fix applied, but don't see any improvements on the raspberry. I spent a few hours debugging the communications with strace, but I'm not exactly sure what's happening there, but verified that calling  `rs2_toggle_advanced_mode` seems to always cause the device to be disconnected. I modified rs-depth example to have:

```
    // Get the first connected device
    // The returned object should be released with rs2_delete_device(...)
    rs2_device* dev = rs2_create_device(device_list, 0, &e);
    check_error(e);

    print_device_info(dev);

    rs2_toggle_advanced_mode(dev, 1, &e);
    check_error(e);

    /* Determine depth value corresponding to one meter */
    uint16_t one_meter = (uint16_t)(1.0f / get_depth_unit_value(dev));

    // Create a pipeline to configure, start and stop camera streaming
    // The returned object should be released with rs2_delete_pipeline(...)
    rs2_pipeline* pipeline =  rs2_create_pipeline(ctx, &e);
    check_error(e);
```

and stepped through the code, and made the following findings.

After executing
```
send_receive(encode_command(ds::fw_cmd::EN_ADV, enable));
```

in `/home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:55`, and then further

```
auto res = _hw_monitor->send(input);
```

in  `librealsense/src/ds5/advanced_mode/advanced_mode.cpp:781`, the first error is printed in kern.log

```
Jul 18 12:03:06 raspberrypi kernel: [  295.196397] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

Then, in in `/home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:56`,

```
send_receive(encode_command(ds::fw_cmd::HWRST));
```

eventually calls the following:

```
(gdb) bt
#0  librealsense::uvc_sensor::invoke_powered<librealsense::locked_transfer::send_receive(std::vector<unsigned char, std::allocator<unsigned char> > const&, int, bool)::{lambda(librealsense::platform::uvc_device&)#2}>(librealsense::locked_transfer::send_receive(std::vector<unsigned char, std::allocator<unsigned char> > const&, int, bool)::{lambda(librealsense::platform::uvc_device&)#2}) (this=0x12efa4, action=...) at /home/pi/librealsense/src/sensor.h:227
#1  0xb62cc65c in librealsense::locked_transfer::send_receive (this=0x131668,
    data=std::vector of length 24, capacity 24 = {...}, timeout_ms=5000,
    require_response=true) at /home/pi/librealsense/src/hw-monitor.h:219
#2  0xb62c847c in librealsense::hw_monitor::send (this=0x11d04c,
    data=std::vector of length 24, capacity 24 = {...})
    at /home/pi/librealsense/src/hw-monitor.cpp:117
#3  0xb5e86cec in librealsense::ds5_advanced_mode_base::send_receive (this=0x12ed44,
    input=std::vector of length 24, capacity 1024 = {...})
    at /home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:781
#4  0xb5e828fc in librealsense::ds5_advanced_mode_base::toggle_advanced_mode (this=0x12ed44,
    enable=true) at /home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:55
#5  0xb5e6f9bc in rs2_toggle_advanced_mode (dev=0x134410, enable=1, error=0xbefff40c)
    at /home/pi/librealsense/src/ds5/advanced_mode/rs_advanced_mode.cpp:46
#6  0x000110dc in main () at rs-depth.c:102
```

and kern.log shows the device disconnecting and reconnecting.

After this, the device may stream a few odd frames, but then the connection dies. If the advanced mode is not toggled, the device appears to stream fine.

filipematosinov, 2019-08-01:
Anyone with any luck? I am also failing using RPi 4 :(

moorage, 2019-08-07:
Did either @filipematosinov  or @SirDifferential  or anyone try installing ubuntu on the RP4 to accomplish this, or would this add even more headaches? :)

filipematosinov, 2019-08-08:
I got librealsense and pyrealsense2 working on RPi4 with Rapsbian Buster :) I don't know which version of Ubuntu is available for the RPi4, but the MATE is not yet.

closed 2019-08-20 by dorodnic

schoningerJim, 2019-08-22:
> I got librealsense and pyrealsense2 working on RPi4 with Rapsbian Buster :) I don't know which version of Ubuntu is available for the RPi4, but the MATE is not yet.

Hi,
Could you share the steps you did?
I would like to build the lib for my raspberry pi 4 too

Thanks in advance.

filipematosinov, 2019-08-22:
Using Raspbian Buster follow the normal instuctions for Raspbian with a slight modification. 
Check here https://github.com/IntelRealSense/librealsense/issues/4565#issuecomment-518359584

GOBish, 2019-09-26:
I am trying to set up a PI 4 for use with a D435 and T265 and having a lot of trouble - I can't even get the realsense viewer running.
I've been following the guide for the pi3 found here
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md
as suggested by @filipematosinov  (using raspian buster as well).
I get an error at the beginning steps that the following packages are not found:

E: Unable to locate package libdrm-amdgpu1-dbg
E: Unable to locate package libdrm-exynos1-dbg
E: Unable to locate package libdrm-freedreno1-dbg
E: Unable to locate package libdrm-nouveau2-dbg
E: Unable to locate package libdrm-omap1-dbg
E: Unable to locate package libdrm-radeon1-dbg
E: Unable to locate package libdrm-tegra0-dbg
E: Unable to locate package libdrm2-dbg

when I try to run to run realsense-viewer at the end of all the steps, I get
zsh: command not found: realsense-viewer

I would appreciate any help!

GOBish, 2019-09-27:
@MartyG-RealSense  -- do you recommend for now just using an upsquared board? I've been having a really hard time trying to get the raspberry pi 4 to work.

MartyG-RealSense, 2019-09-27:
Up Squared boards certainly are compatible with 400 Series cameras.  Up use the board in their own RealSense robotics maker kit.

https://support.intelrealsense.com/hc/en-us/community/posts/360033532514-RealSense-and-the-AAEON-UP-Squared-Robomaker-Dev-Kit-

GOBish, 2019-09-27:
Thanks @MartyG-RealSense  - I have seen that kit - I was just wondering what your take is on the probability of the raspberry pi 4 becoming viable for realsense.

MartyG-RealSense, 2019-09-27:
I would have thought that installation of Librealsense on Pi4 would be possible using  the **libuvc-backend** installation method, as it was made for situations where you want to avoid kernel patching and the normal Librealsense installation instructions do not work.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

If that does not work, specialist installation guides for certain platforms tend to emerge over time, like they did for Pi3 and for Jetson TX2.  Typically they are worked out by RealSense community members and published for the benefit of others.

GOBish, 2019-09-27:
@MartyG-RealSense thanks as always for the advice and quick replies. I appreciate it.

jgwinner, 2019-10-20:
TL;DR No one's gotten it to work?

Whoops, I did a search for my compile problem and didn't quite catch this before opening another bug. I've got the same problem. Will look through the resolution here to see if it'll work.

I tried the PineH64 board before getting the Pi 4, but THAT was a cluster#uck. Amrbian formally said they wouldn't support the Pine H64 board (and were rather unprofessional about it too).

== John ==

nikhilkoditala, 2019-10-21:
I got it working with a single realsense camera. If I add two d415's to rpi4 I am experiencing power issues.

MartyG-RealSense, 2019-10-21:
When power issues occur with using cameras with single-board computing devices, a mains-powered USB hub can provide stability.  I appreciate that this is not ideal for projects that require mobility though, such as moving robots.

nikhilkoditala, 2019-10-21:
@MartyG-RealSense  I didn't use a powered usb hub, but I used extention cables which have power outlets to connect the cameras. But, the rpi4 is getting backpowered and the keyboard and mouse suddenly stopped working. Is there any way to solve this issue?

MartyG-RealSense, 2019-10-21:
You could check that the Pi board itself is not overheating.  A way to test this may be to touch the camera casing and see if it is hot.

masynthetic, 2019-11-15:
@nikhilkoditala could you share your method for getting the 415 running on Pi 4? Struggling to find any working install method.

SirDifferential, 2019-11-15:
With the latest librealsense master, my Pi 4 runs the 415 and 435 cameras without issues as long you build the code with `cmake .. -DFORCE_RSUSB_BACKEND=ON`. They renamed LIBUVC into RSUSB.

masynthetic, 2019-11-15:
> With the latest librealsense master, my Pi 4 runs the 415 and 435 cameras without issues as long you build the code with `cmake .. -DFORCE_RSUSB_BACKEND=ON`. They renamed LIBUVC into RSUSB. There's still a few oddities, but they are too technical / specific to really go into before I know more.

thank you, do I follow the [LibUVC-backend installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)?

SirDifferential, 2019-11-15:
Yeah, I'd guess those instructions work. They still use the old `FORCE_LIBUVC` flag, but the cmake file translates that into `FORCE_RSUSB_BACKEND`.

masynthetic, 2019-11-15:
> Yeah, I'd guess those instructions work. They still use the old `FORCE_LIBUVC` flag, but the cmake file translates that into `FORCE_RSUSB_BACKEND`.

Thanks I got a good install just following the linux instructions with 
<cmake .. -DFORCE_RSUSB_BACKEND=ON>

are you using OpenGL?

SirDifferential, 2019-11-18:
> 
> are you using OpenGL?
> 

No, I run headless without a graphical window.

> I am in /tools/realsense-viewer and I enter `./realsense-viewer` and get command not found,

Does the file exist? Most likely your compilation failed before the executable files were created.

> also tried from /tools, `./realsense-viewer.cpp` gives permission denied and `sudo ./realsense-viewer.cpp` also gives command not found ?

You can't execute `.cpp` files. They are source code you need to compile into executables.

masynthetic, 2019-11-18:
> No, I run headless without a graphical window.
>
>
> Does the file exist? Most likely your compilation failed before the executable files were created.
>
>
> You can't execute `.cpp` files. They are source [...]

Thank you again sir. It does not look like I have executable so I am assuming you're right. I'll give uninstall/reinstall a shot

hr0109, 2019-12-07:
> With the latest librealsense master, my Pi 4 runs the 415 and 435 cameras without issues as long you build the code with `cmake .. -DFORCE_RSUSB_BACKEND=ON`. They renamed LIBUVC into RSUSB. There's still a few oddities, but they are too technical / specific to really go into before I know more.

Thanks！You are really my saviour.

waspinator, 2020-01-12:
Raspberry Pi 4 4GB, Raspbian GNU/Linux 10 (buster) 4.19.75-v7l+

after trying to run a modified [libuvc_installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) with [line 45 ](https://github.com/IntelRealSense/librealsense/blob/v2.31.0/scripts/libuvc_installation.sh#L45) modified to 

```
cmake ../ -DFORCE_LIBUVC=true -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release
```

 I get this error during compilation.

```
[ 58%] Linking CXX shared library librealsense2-gl.so
[ 58%] Built target realsense2-gl
make: *** [Makefile:130: all] Error 2
```

Am I changing the wrong option?

MartyG-RealSense, 2020-01-12:
Here is an example of a CMake build command for the new RUSB system that was introduced in SDK version 2.30.0 onwards.

cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

I note that in your command, you are setting -DFORCE_RSUSB_BACKEND to "On" instead of "True".  Also, it looks as though you could leave out -DFORCE_LIBUVC=true

waspinator, 2020-01-12:
Thanks @MartyG-RealSense for the quick response. I updated line 45 to match your suggestion, but after re-running `libuvc_installation.sh` I got the same error as before. 

I had to add 

```
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic") 
```

to the very top of `librealsense_build/librealsense-master/CMakeLists.txt` after manually running each command in `libuvc_installation.sh` up to line 44, but before line 45. This time it finished compiling and I was able to install and successfully run and view data in realsense-viewer.

MartyG-RealSense, 2020-01-12:
I'm very glad you found a solution.  Thanks for sharing your method with the community!

GOBish, 2020-01-13:
Thanks everyone for the responses. I thought the word was that with the new version of the realsense install, we wouldn't have to use the whole -latomic runaround for the pi4.

GOBish, 2020-01-18:
Doesn't work for me. I have a 4GB version RPi 4, what are you running with?

waspinator, 2020-01-18:
same. using the standard raspbian image from September 2019.