realsenseai/librealsense #4274: depth_example not working with Matlab 2018a on Windows 7

filartrix, 2019-06-24:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415               | 
| Firmware Version                   | 05.11.06.250   | 
| Operating System & Version |   Windows 7   | 
| Platform                                 | PC |
| SDK Version                           |  2.16.1                       | 
| Language                            | Matlab    2018a   | 

### Issue Description
I tried "depth_example" with Matlab, but it does not appear to work.
I've installed realsense SDK fow Win7 and copie the +realsense  package in my Matlab home folder, then simply executed:

` realsense.depth_example`
and got as result:
```
Error using realsense.colorizer/colorize (line 17)
Expected input number 2, depth, to be a depth_frame

Error in realsense.depth_example (line 26)
    color = colorizer.colorize(depth);
```
I can correctly get the Camera name from Matlab

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

@filartrix Noticed that you're using an old version SDK 2.16.1. Could you please try the latest release in https://github.com/IntelRealSense/librealsense/releases to see if you still have issue? Thanks!

RealSenseCustomerSupport, 2019-08-07:
----------------------------------------------

@filartrix Could you please update if you still have issue? Thanks!

filartrix, 2019-08-08:
Hello, thanks for the suggestion, I tried with latest release, bu the issue on depth frames is still there
I checked that I can correctly get a color frame, but not a depth frame. the example code I used is:

 `    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    pointcloud = realsense.pointcloud();
    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();
    % Obtain frames from a streaming device
    fs = pipe.wait_for_frames();
    % Select depth frame
    depth = fs.get_depth_frame();
    color = fs.get_color_frame();
`
if I call depth.logical() result is zero (false)
but for color.logical(), result is 1, and I can use the methods to get frame information and data.
regards

RealSenseCustomerSupport, 2019-08-16:
----------------------------------------------

@filartrix Can your camera get depth normally with RealSenseViewer(https://github.com/IntelRealSense/librealsense/releases/download/v2.25.0/Intel.RealSense.Viewer.exe)?

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

@filartrix Any update for this issue? Is your Win7 system 64bit or 32bit?

RealSenseCustomerSupport, 2019-09-16:
----------------------------------------------

@filartrix Sorry that we didn't get feedback from you for a month. Will close this issue at this point. Please feel free to create new ticket if you still have questions or issues. Thanks!

closed 2019-09-16 by RealSenseCustomerSupport