realsenseai/librealsense #4277: RealSense SDk and building Dependencies

HadeelSurougi, 2019-06-24:
Hello,

I just got RealSense Tracking camera T265 and I installed the SDK and the sample examples. However, I am trying to build my own C++ project from the scratch in Visual Studio 2019  and I tried to add the RealSense library to my project and link it. But, I can't include the library, the program can't define it yet. So could you help me in providing the right steps in building all the external dependencies of RealSense and should I use the CMake in that?

Note: It is the first time for me to deal with the Visual Studio and CMake

Thank you

dorodnic, 2019-06-24:
All you need is `realsense2.lib`, `realsense2.dll` and `include` folder, all provided inside the Windows Installer. You configure linker and include directories in a standard way, and place `realsense2.dll` near your application. Alternatively, the SDK is installing `C:\Program Files (x86)\Intel RealSense SDK 2.0\intel.realsense.props` file that when included will configure your C++ project to use the DLL from program files. [Here](https://dorodnic.com/blog/2014/03/20/visual-studio-macros/) is somewhat outdated but yet relevant guide of using PROP files in Visual Studio in general.

radfordi, 2019-06-25:
The standard `cmake` technique is to add the location of `librealsense` to your `CMAKE_PREFIX_PATH` with something like this (run in your build directory):
```sh
cmake -DCMAKE_PREFIX_PATH="C:\Program Files (x86)\Intel RealSense SDK 2.0" .
```
and then to ask `cmake` to find `librealsense` using that path in your `CMakeLists.txt` like so
```cmake
find_package(realsense2)

target_link_libraries(${PROJECT_NAME} PRIVATE realsense2::realsense2)
```
this should deal with the include paths, finding the DLL, etc. automatically.

HadeelRS, 2019-07-02:
Dorodnic and Radfordi thank you so much for your help .. I'll try.

RealSenseCustomerSupport, 2019-07-18:
----------------------------------------------

Hi @HadeelRS​,

Any update? Is the issue resolved per the suggestions?

closed 2019-07-22 by dorodnic

monson-li, 2024-08-22:
> The standard `cmake` technique is to add the location of `librealsense` to your `CMAKE_PREFIX_PATH` with something like this (run in your build directory):
> 
> ```shell
> cmake -DCMAKE_PREFIX_PATH="C:\Program Files (x86)\Intel RealSense SDK 2.0" .
> ```
> 
> and then to ask `cmake` to find `librealsense` using that path in your `CMakeLists.txt` like so
> 
> ```cmake
> find_package(realsense2)
> 
> target_link_libraries(${PROJECT_NAME} PRIVATE realsense2::realsense2)
> ```
> 
> this should deal with the include paths, finding the DLL, etc. automatically.

Hello, radfordi.  Thanks for your answer! What is the meaning of "realsense2::realsense2"?