realsenseai/librealsense #4278: Issues with GLFW3

ClementDardenne, 2019-06-24:
Hi,

I tried for hours to build librealsense2 under Windows 10 (with cmake & VS 2015/2017).
My goal is to use rs-kinfu module : https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

My problem is related to GLFW3. I can generate my Project with cmake, but if I try to build rs-kinfu with Visual Studio 2015/2017 in Release, I have : 

`C:\..\examples\example.hpp(9): fatal error C1083: Cannot open include file: 'GLFW/glfw3.h': No such file or directory`

Note I have this message in cmake : "GLFW 3.3 not found; using internal version" (so it should use the default version in third-party)

Then, I tried to install my own GLFW : https://www.glfw.org/download.html (source). My problem is now in cmake, I can't link glfw3_DIR (I tried to specify : C:\...\glfw-3.3\src, which seems to be the correct folder), but it's not working.

Then, I followed the recommandations of : https://github.com/IntelRealSense/librealsense/issues/3304

So I linked in Visual Studio glfw3.lib (pre-compiled binaries from https://www.glfw.org/download.html) and I add the include directories ...\glfw-3.3.bin.WIN64\include\GLFW (containing glfw3.h)

I still got errors : 
[image]

Someone could help me to find a solution ? 

Thank you in advance,

Clément

ClementDardenne, 2019-06-25:
Hi again,

I finally fix most of the linking errors by adding OpenGL32.Lib and GlU32.Lib in addition to glfw3.lib in VS2017. 

I have a last linking error : 

`error LNK2019: unresolved external symbol GUID_DEVINTERFACE_USB_DEVICE referenced in function "bool __cdecl librealsense::platform::get_usb_descriptors(unsigned short,unsigned short,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &,enum librealsense::platform::usb_spec &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)" (?get_usb_descriptors@platform@librealsense@@YA_NGGAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEAV34@AEAW4usb_spec@12@1@Z)
1>C:\Users\Clement.Dardenne\Documents\lib\librealsense\build\Release\realsense2.dll : fatal error LNK1120: 1 unresolved externals`

I am still googling the issue, but it seems I can't find the lib to add this time. Any idea ?

Thanks,

Clément

ClementDardenne, 2019-06-26:
Finally, I find a solution by changing :

```
#ifndef WITH_TRACKING
DEFINE_GUID(GUID_DEVINTERFACE_USB_DEVICE, 0xA5DCBF10L, 0x6530, 0x11D2, 0x90, 0x1F, 0x00, 0xC0, 0x4F, 0xB9, 0x51, 0xED);
#endif
```

by

`DEFINE_GUID(GUID_DEVINTERFACE_USB_DEVICE, 0xA5DCBF10L, 0x6530, 0x11D2, 0x90, 0x1F, 0x00, 0xC0, 0x4F, 0xB9, 0x51, 0xED);`

in win-helpers.cpp

closed 2019-06-26 by ClementDardenne

Drq13112, 2024-10-23:
@ClementDardenne Hi, did you test the kinfu code??