realsenseai/librealsense #4281: Possible bug in capturing: Frame didn't arrive within 5000 / Error occured during execution of the processing block!

pabloalcain, 2019-06-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Debian 8  | 
| Kernel Version (Linux Only)    |  4.9.0-8-amd64                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2                          | 
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description

Hello everyone. I have been working lately with the python bindings of the realsense2 library with the D415 model in Linux, and found something that might be a bug (or likely underdocumented error) when accumulating frames.

 

When I append the outputs from `pipeline.wait_for_frames()` to different python lists, it looks like it blocks the depth or color frame object and after 16 frames of appending it throws an error "Frame didn't arrive within 5000". If we try to align to the color frame, the error is "Error occured during execution of the processing block!".

However, this is easily overcome when appending a copy of said frames, which looks like a good solution and not too costly performance-wise. Before thinking of any PR to the main repo, I wanted to know whether this behavior was expected.

I am yet to understand completely the nature of the bug, but here I leave you the gists that trigger both errors, and the `copy` workaround.

If it helps, I found is not time-related but rather number of frames related, and it also is not dependent (I haven't checked all the range though) on the resolution.

Pablo

 

https://gist.github.com/pabloalcain/65610225deb0ae9335e66e1a6c755c57

RealSenseCustomerSupport, 2019-08-27:
----------------------------------------------

@pabloalcain Sorry for the late reply. What's the SDK version when you get this issue? Did you get chance to try the latest SDK version and FW to see if you still get the issue? Thanks!

RealSenseCustomerSupport, 2019-09-10:
----------------------------------------------

@pabloalcain Did you still have issue with the latest SDK and FW? Could you please feedback?Thanks!

RealSenseCustomerSupport, 2019-09-23:
----------------------------------------------

@pabloalcain Thanks for the feedback! Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@pabloalcain Any update from your side now? Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@pabloalcain Did you get the HW now to test? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@pabloalcain Sorry that we didn't get response from you for weeks. Will close this if no other questions. Please feel free to create new one if you still have any other questions/issues. Thanks!

closed 2019-10-25 by RealSenseCustomerSupport