realsenseai/librealsense #4282: Map stitching with T265 for large scale application ?

ParadoxRobotics, 2019-06-24:
I would like to perform VIO using the T265 on a 1km road.
Since the T265 have limited memory, is it possible to perform map stitching with load/save internal map on a external computer ?
Also if it is possible, how can we know the limit of the map ? Does the T265 internal map memory have a maximum flag ?

radfordi, 2019-06-25:
How large a map you need depends on the scale of your scene.  If you are high up in drone with a downward facing T265 you might be able to see the same scene while moving over large distances so the map size would be small with respect to the distance covered, compared flying low over the ground which would eat up your map in relatively short distance.

The T265 map has a fixed size for now.  We are currently focusing on how to increase this size rather than how to, for example, tile/stitch map sections as the API for the former is more clear.

Can you explain a bit about what you are trying to?  Have you tried T265 on your driving example?  Is it a looping 1km road?  Do you have a saved map you could post?

ParadoxRobotics, 2019-06-25:
I am currently searching for an embedded SLAM approach for a forest robot (inspection task) using low computer ressource (Jetson nano or the new RPI4).
I hesitate with a classical approach using a SLAM algorithm (orb-slam or other) on a normal computer or a low-power approach using the T265.
The travel distance is approximately 900m on a forest track.
I assume the current implementation of the T265 is in most case for indoor navigation on a relatively small scale (2 to 3 rooms of a house depending on the complexity of the scene).

radfordi, 2019-06-25:
You are correct that we have mostly tested indoors, but we have tested on drones and driven in cars outdoors.  Maybe forests will work? 🤷‍♂  We'd love to hear if they do!

ParadoxRobotics, 2019-06-26:
I hope other people will also try the T265 in large / long scale condition ^^

Sttrr, 2019-08-14:
> I would like to perform VIO using the T265 on a 1km road.
> Since the T265 have limited memory, is it possible to perform map stitching with load/save internal map on a external computer ?
> Also if it [...]

> I would like to perform VIO using the T265 on a 1km road.
> Since the T265 have limited memory, is it possible to perform map stitching with load/save internal map on a external computer ?
> Also if it [...]

how to save a pointcloud map using t256??,thanks

radfordi, 2019-08-14:
@Sttrr, you'll need to pair the T265 with a depth camera (say a D435) in order to get a point cloud.  See this [`tracking-and-depth` example](https://github.com/IntelRealSense/librealsense/tree/master/examples/tracking-and-depth).

giovannicairoli, 2019-09-05:
Hello @ParadoxRobotics,

I'm not super expert in programming and I need a little help. I'm doing something similar to your application in an indoor context. I already have a T265 and a D435. Could you please help me in understanding how to use them in order to save a map, import export it and re-locate my device in a known context? Samples or Documents? I did't find any document and I don't know where to start.

Thnak you very much for your help.

RealSenseSupport, 2020-09-11:
Thank you for highlighting the need for map stitching and larger map sizes using T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

ParadoxRobotics, 2020-09-15:
So if I understand the T65 will only be good on small scale SLAM without any expansion.

RealSenseSupport, 2020-09-15:
We have moved our focus to our next generation of products and consequently, we will not be addressing issues/new features in the T265.

closed 2020-09-17 by RealSenseSupport