realsenseai/librealsense #4283: T265 Device not connected on Raspian

etharooni, 2019-06-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |  Raspian Stretch Lite | 
| Kernel Version (Linux Only)    |  4.19.42-v7+ | 
| Platform                                 | Raspberry Pi |
| SDK Version                            | 2.23.0 d6f6be | 
| Language                               | Python |

### Issue Description
I am unable to get the USB drivers for my T265 camera to recognize the device when using Raspian on a Raspberry Pi 3 A+.  I have followed the instructions from here for getting the device set up with librealsense: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

Here is the output of lsusb:
```
pi@raspberrypi:~/librealsense$ lsusb
Bus 001 Device 002: ID 03e7:2150
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Here is the output of dmesg:
```
[  892.666579] usb 1-1: new high-speed USB device number 2 using dwc_otg
[  892.666729] Indeed it is in host mode hprt0 = 00001101
[  892.907229] usb 1-1: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[  892.907241] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  892.907251] usb 1-1: Product: Movidius MA2X5X
[  892.907260] usb 1-1: Manufacturer: Movidius Ltd.
[  892.907268] usb 1-1: SerialNumber: 03e72150
```

hello-realsense example:
```
pi@raspberrypi:~/librealsense/examples/hello-realsense$ ./a.out
RealSense error calling rs2_pipeline_start(pipe:0x85bbd8):
    No device connected

```
Any help getting the device to show up would be very appreciated!

ev-mp, 2019-06-25:
@etharooni hello,
T265 device appears in the HW log:
>Bus 001 Device 002: ID 03e7:2150

So hw-related issue can be safely ruled out.

Please check the following:
- Internet access during compilation is required for T265 support (see #4285)
- Librealsense's `udev-rules` are properly [applied](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites):
>./scripts/setup_udev_rules.sh

etharooni, 2019-06-25:
@ev-mp 
I just re-ran the scripts and it appears to have the same symptoms.  

I have also had internet access during the entire compilation process - is there a way to check to see if T265 support has been compiled?

ev-mp, 2019-06-25:
@etharooni , during CMake configuration make sure to request T265 module to be built-in:
>cmake ../  -DBUILD_WITH_TM2=true  -D..... 

Then follow the CMake log for TM2-related messages - you should see report whether internet connection was established and the module was properly configured

etharooni, 2019-06-25:
Interesting. 

`-- Checking internet connection...
-- Failed to identify Internet connection, disabling BUILD_WITH_TM2`

I changed the timeout in the connectivity check by changing the line 
`file(DOWNLOAD "http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/connectivity_check" "${CMAKE_CURRENT_SOURCE_DIR}/connectivity_check" SHOW_PROGRESS TIMEOUT 5 STATUS status)`
in file CMake/tm2_connectivity_check.cmake to
`file(DOWNLOAD "http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/connectivity_check" "${CMAKE_CURRENT_SOURCE_DIR}/connectivity_check" SHOW_PROGRESS TIMEOUT 15 STATUS status)`
and it appears to be building with TM2.  I'll report back once it finishes building.

etharooni, 2019-06-26:
Success!  Suggestion: maybe timeout should be longer.

closed 2019-06-26 by etharooni