realsenseai/librealsense #4284: RealSense Viewer on Nano not connecting to T265.

tmichals, 2019-06-24:
| | |
|---|---|
|**librealsense**|2.21.0 RELEASE|
|**OS**|Ubuntu 18|

Unable to get the RealSense Viewer to display the T265... can not add the T265.   I have worked through the similar related github issues that talk about the same problem  ....but not able to get T265 working .   The same T265 camera works perfectly on my windows10 PC with the RealSense viewer so the camera is good.     I am using a nvidia nano and installing the RealSense SDK per instructions found on JetsonHacks( https://www.youtube.com/watch?v=NonBlt-wKCI ) .  The Jetson install completes with no obvious errors..   I also have a d435 camera and it connects perfectly to the RealSense Viewer on the nano.   I have been working with JetsonHacks and we have a question about the firmware... the T265 is unable to connect to the web site (http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/connectivity_check) to install the proper firmware.   When I try to connect using a web browser on the nano it says  403 error forbidden access.  I get the same error from my windows10 PC.   Perhaps the web site is not designed to test connectivity directly through a web browser... but JetsonHacks believes it is something else.   Is there a way to load the t265 firmware similar to the d435?  How can I confirm that the t265 is loading the firmware correctly?  Is there a set of scripts that I can run to test that the firmware is loading on the t265 for the nano and Ubuntu18?

mickeyouyou, 2019-06-25:
It's same situation to me. 

ubuntu 18.04 T265

tmichals, 2019-06-25:
Yes I have seen the same issue as mickeyouyou .  Below the lsusb shows Bus 001 Device 007: ID 03e7:2150 is connected.

lcs@lcs:~$ lsusb
Bus 002 Device 004: ID 2109:8110 VIA Labs, Inc. Hub
Bus 002 Device 003: ID 2109:8110 VIA Labs, Inc. Hub
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 03e7:2150  
Bus 001 Device 006: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 005: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 004: ID 046d:c06a Logitech, Inc. USB Optical Mouse
Bus 001 Device 003: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lcs@lcs:/usr/local/bin$ rs-pose
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55a5b28440, config:0x55a5b0afc0):
    No device connected

dorodnic, 2019-06-25:
We need to check and update this thread, hopefully within couple of days. As it stands Nano is not part of our official test matrix, but we understand the importance of the platform.

tmichals, 2019-07-02:
Thanks dorodnic.   Hopefully we can get this issue figured out.

tmichals, 2019-07-16:
Still waiting....

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@tmichals Is the issue still existing? And can you access the links below?
http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.1.0.279/target-0.1.0.279.mvcmd
http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/app/2.0.19.271/central_app-2.0.19.271.bin
http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/bl/1.0.1.112/central_bl-1.0.1.112.bin

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@tmichals Any update from you side? Thanks!

tmichals, 2019-11-07:
Yes, I was able to get the T265 to connect to the RealSense Viewer by following JetsonHacks T265 Nano install and adding  "-DBUILD_WITH_TM2=true" flag ....   cmake ../ -DBUILD_WITH_TM2=true  

For various other reasons when I am working on a roslaunch for realsense, I may loose the T265 connection again .... but rebuilding with flag fixes that problem.   T265 works as long as I do not create a roslaunch error.

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

@tmichals Do you mean that sometimes you still get T265 disconnection issue if use roslaunch?  Which launch file did you run when get the issue?

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@tmichals  Could you please update? Thanks!

tmichals, 2019-11-22:
I should be able to look at it by this weekend.

tmichals, 2019-12-01:
At times when I am not careful on shutting down the T265 service, like just powering off Nano, or not setting up the ROSCORE correctly and trying to start T265.... the camera will just disappear...until I rebuild with the sudo cmake ../ -Build_with_TM2=true.

Below is what it looks like when this happens....
____________________________________________________________________________________________________ 
lcs@nano:~/catkin_ws$ roslaunch realsense2_camera camera:=cam_2 rs_t265.launch serial_no:=845412110091
... logging to /home/lcs/.ros/log/30ecd4dc-e2e5-11e9-aa1e-00044be40a0e/roslaunch-nano-7006.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.72.20.56:36801/

SUMMARY
========

PARAMETERS
 * /cam_2/realsense2_camera/accel_fps: 62
 * /cam_2/realsense2_camera/accel_frame_id: cam_2_accel_frame
 * /cam_2/realsense2_camera/accel_optical_frame_id: cam_2_accel_optic...
 * /cam_2/realsense2_camera/align_depth: False
 * /cam_2/realsense2_camera/aligned_depth_to_color_frame_id: cam_2_aligned_dep...
 * /cam_2/realsense2_camera/aligned_depth_to_fisheye1_frame_id: cam_2_aligned_dep...
 * /cam_2/realsense2_camera/aligned_depth_to_fisheye2_frame_id: cam_2_aligned_dep...
 * /cam_2/realsense2_camera/aligned_depth_to_fisheye_frame_id: cam_2_aligned_dep...
 * /cam_2/realsense2_camera/aligned_depth_to_infra1_frame_id: cam_2_aligned_dep...
 * /cam_2/realsense2_camera/aligned_depth_to_infra2_frame_id: cam_2_aligned_dep...
 * /cam_2/realsense2_camera/allow_no_texture_points: False
 * /cam_2/realsense2_camera/base_frame_id: cam_2_link
 * /cam_2/realsense2_camera/calib_odom_file:
 * /cam_2/realsense2_camera/clip_distance: -1.0
 * /cam_2/realsense2_camera/color_fps: 30
 * /cam_2/realsense2_camera/color_frame_id: cam_2_color_frame
 * /cam_2/realsense2_camera/color_height: 480
 * /cam_2/realsense2_camera/color_optical_frame_id: cam_2_color_optic...
 * /cam_2/realsense2_camera/color_width: 640
 * /cam_2/realsense2_camera/depth_fps: 30
 * /cam_2/realsense2_camera/depth_frame_id: cam_2_depth_frame
 * /cam_2/realsense2_camera/depth_height: 480
 * /cam_2/realsense2_camera/depth_optical_frame_id: cam_2_depth_optic...
 * /cam_2/realsense2_camera/depth_width: 640
 * /cam_2/realsense2_camera/enable_accel: True
 * /cam_2/realsense2_camera/enable_color: True
 * /cam_2/realsense2_camera/enable_depth: True
 * /cam_2/realsense2_camera/enable_fisheye1: True
 * /cam_2/realsense2_camera/enable_fisheye2: True
 * /cam_2/realsense2_camera/enable_fisheye: True
 * /cam_2/realsense2_camera/enable_gyro: True
 * /cam_2/realsense2_camera/enable_infra1: True
 * /cam_2/realsense2_camera/enable_infra2: True
 * /cam_2/realsense2_camera/enable_pointcloud: False
 * /cam_2/realsense2_camera/enable_sync: False
 * /cam_2/realsense2_camera/filters:
 * /cam_2/realsense2_camera/fisheye1_frame_id: cam_2_fisheye1_frame
 * /cam_2/realsense2_camera/fisheye1_optical_frame_id: cam_2_fisheye1_op...
 * /cam_2/realsense2_camera/fisheye2_frame_id: cam_2_fisheye2_frame
 * /cam_2/realsense2_camera/fisheye2_optical_frame_id: cam_2_fisheye2_op...
 * /cam_2/realsense2_camera/fisheye_fps: 30
 * /cam_2/realsense2_camera/fisheye_frame_id: cam_2_fisheye_frame
 * /cam_2/realsense2_camera/fisheye_height: 800
 * /cam_2/realsense2_camera/fisheye_optical_frame_id: cam_2_fisheye_opt...
 * /cam_2/realsense2_camera/fisheye_width: 848
 * /cam_2/realsense2_camera/gyro_fps: 200
 * /cam_2/realsense2_camera/gyro_frame_id: cam_2_gyro_frame
 * /cam_2/realsense2_camera/gyro_optical_frame_id: cam_2_gyro_optica...
 * /cam_2/realsense2_camera/imu_optical_frame_id: cam_2_imu_optical...
 * /cam_2/realsense2_camera/infra1_frame_id: cam_2_infra1_frame
 * /cam_2/realsense2_camera/infra1_optical_frame_id: cam_2_infra1_opti...
 * /cam_2/realsense2_camera/infra2_frame_id: cam_2_infra2_frame
 * /cam_2/realsense2_camera/infra2_optical_frame_id: cam_2_infra2_opti...
 * /cam_2/realsense2_camera/infra_fps: 30
 * /cam_2/realsense2_camera/infra_height: 480
 * /cam_2/realsense2_camera/infra_width: 640
 * /cam_2/realsense2_camera/initial_reset: False
 * /cam_2/realsense2_camera/json_file_path:
 * /cam_2/realsense2_camera/linear_accel_cov: 0.01
 * /cam_2/realsense2_camera/odom_frame_id: cam_2_odom_frame
 * /cam_2/realsense2_camera/pointcloud_texture_index: 0
 * /cam_2/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /cam_2/realsense2_camera/pose_frame_id: cam_2_pose_frame
 * /cam_2/realsense2_camera/pose_optical_frame_id: cam_2_pose_optica...
 * /cam_2/realsense2_camera/publish_odom_tf: True
 * /cam_2/realsense2_camera/rosbag_filename:
 * /cam_2/realsense2_camera/serial_no: 845412110091
 * /cam_2/realsense2_camera/topic_odom_in: cam_2/odom_in
 * /cam_2/realsense2_camera/unite_imu_method:
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /cam_2/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://10.72.20.56:5800

process[cam_2/realsense2_camera_manager-1]: started with pid [7043]
process[cam_2/realsense2_camera-2]: started with pid [7044]
[ INFO] [1569780996.399808982]: Initializing nodelet with 4 worker threads.
[ INFO] [1569780996.981739242]: RealSense ROS v2.2.7
[ INFO] [1569780996.981859034]: Running with LibRealSense v2.24.0
[ERROR] [1569780997.451805596]: The requested device with serial number 845412110091 is NOT found. Will Try again.
[ERROR] [1569781003.543244344]: The requested device with serial number 845412110091 is NOT found. Will Try again.
[ERROR] [1569781009.630988821]: The requested device with serial number 845412110091 is NOT found. Will Try again.
[ERROR] [1569781015.722622360]: The requested device with serial number 845412110091 is NOT found. Will Try again.

[image]

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

@tmichals We couldn't reproduce the issue here with SDK v.2.31 and ROS v2.2.11 on the Nano. Could you please try if you still get the issue with the latest version? Thanks!

tmichals, 2019-12-17:
Okay.  I will let you know.

tmichals, 2019-12-26:
I installed the latest from you and JetsonHacks ...  I flashed Nano with JetPack 4.3 - 32.3.1 and the Intel RealSense Viewer v2.31.0 ... along with ROS Melodic, and Intel RealSense ROS Wrappers.   
The T265 was selectable from the realsense-viewer!   That has never happed before!    It appears fixed!

closed 2019-12-26 by tmichals