realsenseai/librealsense #4285: T265 No device connected, please connect a RealSense device

mickeyouyou, 2019-06-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I have checked by approch in issue https://github.com/IntelRealSense/librealsense/issues/3488
But I use https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h api . `get_a_realsense_device ` there is error :

```
No device connected, please connect a RealSense device
```

ev-mp, 2019-06-25:
@mickeyouyou , T265 build requires internet connection for firmware download, and in absence of it the T265 device compilation is skipped. This event is reported to the CMake's log but often overlooked.
For the HW part -  can you share `lsusb` output  ?

mickeyouyou, 2019-06-25:
```bash
$ lsusb
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 004: ID 03e7:2150
Bus 001 Device 003: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

ev-mp, 2019-06-25:
The device appears in the list:
>Bus 001 Device 004: ID 03e7:2150

So the issue is probably with the compilation without internet access- please review the CMake log for TM2 event.
Make sure that you have internet access and rebuild librealsense. Proxy might be required for corporate networks. You can also share the logs for review

closed 2019-07-12 by RealSenseCustomerSupport