realsenseai/librealsense #4286: Wavy Point Cloud Issue

pinumber22by7, 2019-06-25:
I am using D435i for getting a pointcloud but I am getting a Wavy(wave type) pattern on the pointcloud and my application(use of the camera) requires detection of a discontinuity which i am unable to get.
any help is appreciated

dorodnic, 2019-06-25:
This type of noise is inherent to D400 technology, but it can be exaggerated by bad calibration, and improved by following [this white-paper](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance)

aPonza, 2019-07-02:
Where can the "Intel OEM Calibration Tool and Chart" mentioned in the paper be found? Both [the provided link](https://www.intelrealsense.com/developers/) and [what I googled](https://downloadcenter.intel.com/product/128255/Intel-RealSense-Depth-Camera-D435) lead to at most Windows 10 stuff (I'm looking for ubuntu, but probably have it) and seemingly no pdfs with stuff to print (this is likely what I'm after).

closed 2019-07-14 by dorodnic