realsenseai/librealsense #4288: Kidnapped T265

JimmyFroost, 2019-06-25:
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 |  
| Operating System & Version | Windows 10  |                                | 
| Platform                                 | PC   | 
| Language                            |  Matlab                    |  
 

Hi

After experimenting T265 for a few months now, I've witnessed the phenomenom where the camera loses its way and finds it back. 
Here is a small example where we see that the trajectory shifts a bit to the left and gets back to normal a few moments later : 
 
[image]
 
Camera was set on a small cart, approximately 50cm above the ground.

My question is : if the  camera is able to relocate itself , what can we do about false coordinates that we get during it's being kidnapped ? 
The confidence is high (=3) but we still have bad positioning ?
Is there any way to fix the bad xyz that have been recorded ?

I cannot find any clear information about this on Intel's website , only a few general and vague stuff [here](https://www.intelrealsense.com/tracking-camera-t265/)
 
I read  :

> Our How-Tos and video tutorials cover everything from basics to advanced techniques.

but I can't find anything. Hopefully I am mistaken and someone can help me ?

ArtlyStyles, 2019-06-25:
how fast was the cart moving?

JimmyFroost, 2019-06-26:
I was walking and pushing the cart quite slowly, slower than a usual walking speed.
I don't need to go fast for the project I'm working on, I'd rather say the slower the better

JimmyFroost, 2019-06-27:
Another day, some other tests, some other kidnapping events !

Let me add this question :
How come the confidence value is still 3 (ie high) when the camera loses its way ? Is there no indicator that allows to understand:

- camera imu is getting "moving state" value
- camera fisheyes see that new elements are coming in display, aka camera is moving

and yet the trajectory is still or speed is almost null ?

PaulKrush, 2019-06-28:
See issue #3516 and PR #4321 

There are two different ways to look at this. Some people want the jump not to happen. I am like you and I want to pose history updated so the jump does not exist in the recorded path.

radfordi, 2019-06-28:
Yes, I think you'll want to disable jumping once #4321 gets merged and the firmware gets updated.

JimmyFroost, 2019-07-01:
Thank you @PaulKrush  & @radfordi , I remember seeing this #4321 indeed.

Hopefully we'll have it in an update soon !

RealSenseCustomerSupport, 2019-08-15:
----------------------------------------------

The "pose jumping" option integrated in 2.25.0.

closed 2019-08-15 by RealSenseCustomerSupport

JimmyFroost, 2019-08-19:
Sorry for "re-opening" this issue, but I'm not sure if this is possible with Matlab ? 
I'm understanding it's not yet.

radfordi, 2019-08-19:
Thanks for pointing out that the Matlab wrappers needed updating @JimmyFroost.  Here's a PR to address that: #4666.