realsenseai/librealsense #4289: Depth image is too dark when using OpenCV

MCSuperior, 2019-06-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-52-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  1.1.2.realsense-1.3.5                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
When I open a pipeline, get a depth stream and make a cv::Mat with it, the resulting depth image is much darker than the one I can see in the RealSense viewer (When using black/white coloring for the depth data). This is bad, as it seems to me that the darker image also contains less information (Is that actually correct?).

[image]
[image]

This is my code:

```
using namespace std;
using namespace cv;

int main() {

    rs2::pipeline pipe;
    rs2::config cfg;

    cfg.enable_all_streams();
    pipe.start(cfg);

    rs2::frameset frames;
    for(int i = 0; i < 30; i++)
    {
        frames = pipe.wait_for_frames();
    }

    rs2::frame depth_frame = frames.get_depth_frame();
    Mat depth(Size(640, 480), CV_16UC1, (void*)depth_frame.get_data(), Mat::AUTO_STEP);
    pipe.stop();

    namedWindow("Depth Image", WINDOW_AUTOSIZE );
    imshow("Depth Image", depth);
    waitKey(0);

    return 0;
}
```

I have the impression that this might be related to the image encoding which I use for the conversion into a Mat object. I also saw some people using CV_8UC1, which totally does not make sense to me and also did not work.

What am I doing wrong? Is this maybe just a displaying error or something?

dorodnic, 2019-06-25:
Well raw depth data is not really intended to be looked at. We provide `rs2::colorizer` class used in all examples that convert depth frame to synthetic RGB visualisation

MCSuperior, 2019-06-26:
Thank you for the answer, that explains why this did not work.
Eventually, I am only interested in obtaining the actual (and most importantly correct) depth at a specified pixel coordinate and I thought that fetching the pixel value in a depth image would work. How do I do it then?

closed 2019-06-27 by MCSuperior