realsenseai/librealsense #4290: Howto get an copy of DepthStream

gdiiorio, 2019-06-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) /   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  C#}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, i have start new project from c# example cs-tutorial-3-processing . I have need to pause, by press a key, the stream image for best object measure.  How-to try to have an copy of DepthStream ? How-to pause the stream ?

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@gdiiorio Please check to see if frame.keep() API below can meet your requirement. Thanks!
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/Frames/Frame.cs#L104-L115
PR link: https://github.com/IntelRealSense/librealsense/pull/1015

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@gdiiorio If no other questions will close this accordingly. Thanks!

closed 2019-10-12 by RealSenseCustomerSupport