realsenseai/librealsense #4291: ExportToPly Gives Incoherent/Garbled Images

NickolasCavagnaro, 2019-06-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version | Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    | 4.15.0-51-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.23.0 | 
| Language                            | nodejs |
| Segment			| others | 

### Issue Description
The NodeJS exportToPLY function does not give accurate pointclouds. As such, it can't be rendered.

I know this feature is still not fully rolled out, but I wanted to export my pointclouds into PLY files using node as I have already written considerable code in node. I was able to get a correct image using the Python wrapper. However, with the node wrapper, when I used Meshlab and three.js to visualize it, all I saw was a garbled mess that did not resemble at all what was supposed to be captured. I have attached an image of what the exported PLY file looks like in Meshlab (for the record, it was a pointcloud of my upper body and face in a big room). Below is my code (in nodejs) that generated this image.

```
const rs = require('./index.js');
const {GLFWWindow} = require('./glfw-window.js');
const {glfw} = require('./glfw-window.js');

const win = new GLFWWindow(1280, 720, 'Capture Stream');

const cfg = new rs.Config();

const pc = new rs.PointCloud();
let points = new rs.Points();

const pipeline = new rs.Pipeline();

const width = 640;
const height = 480;
const fps = 30;

cfg.enableStream(rs.stream.STREAM_DEPTH, 0, width, height, rs.format.FORMAT_Z16, fps);
cfg.enableStream(rs.stream.STREAM_COLOR, 0, width, height, rs.format.FORMAT_RGB8, fps);
cfg.enableStream(rs.stream.STREAM_INFRARED, 1, width, height, rs.format.FORMAT_Y8, fps);
cfg.enableStream(rs.stream.STREAM_INFRARED, 2, width, height, rs.format.FORMAT_Y8, fps);

if (cfg.canResolve(pipeline)) {
    cfg.resolve(pipeline);
}
else {
    throw "Could not resolve the pipeline with given configurations! :(";
}

pipeline.start(cfg);

let ct = 0;
while (!win.shouldWindowClose()) {
    const frames = pipeline.waitForFrames();
    const depthFrame = frames.depthFrame;
    const colorFrame = frames.colorFrame;

    pc.mapTo(colorFrame);

    points = pc.calculate(depthFrame);

    win.beginPaint();
    glfw.drawDepthAndColorAsPointCloud(
        win.window,
        new Uint8Array(points.vertices.buffer),
        points.size,
        new Uint8Array(points.textureCoordinates.buffer),
        colorFrame.data,
        colorFrame.width,
        colorFrame.height,
        'rgb8');
    win.endPaint();

    ct++;
    if (ct === 100) {
        points.exportToPly("/home/nickolas/Farms/textures/FaceCode.ply", colorFrame);
        console.log("Done exporting");
    }
}

pipeline.stop();
pipeline.destroy();
win.destroy();
rs.cleanup();
```

[image]

rupeshchandgude, 2019-07-05:
I am facing same issue. I am using D435 model on ubuntu 16.04.  I am getting below output :
[image]

lramati, 2019-08-20:
Have you tried zooming in on the center of the mesh?
We've seen issues before where mesh viewers (both standalone applications and visualization libraries in various languages) zoom very far out to get some stray points with very far depth values to fit in the viewport, causing the actual data to become way to small to be seen, although it is there.

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@NickolasCavagnaro Did you get chance to try the suggestion from Iramati? Is the issue resolved? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@NickolasCavagnaro No responese from you for weeks. If no other questions will close the ticket accordingly. Thanks!

closed 2019-10-12 by RealSenseCustomerSupport