realsenseai/librealsense #4292: Impossible to access Realsense forum

patrickpoirier51, 2019-06-25:
Sorry to request support here, but it seems that there is no way to access forum ....

My old profile was deleted, so I created a new one , but once I sign in, and try to access forum, it is redirecting to
https://sfederation.intel.com/affwebservices/public/saml2sso?SPID=https://saml.salesforce.com.isvc.forums&ProtocolBinding=urn:oasis:names:tc:SAML:2.0:bindings:HTTP-POST&RelayState=%2Fs%2F

and ends up here:
 https://www.intel.com/content/www/us/en/support/home/v2.html

Please note that I am from Canada

MartyG-RealSense, 2019-06-26:
The Intel Support forums are still available, and their front page can be accessed here:

https://forums.intel.com/s/

The RealSense forum though has moved to a new site at the link below.

https://support.intelrealsense.com/hc/en-us/community/topics

patrickpoirier51, 2019-06-26:
Thanks Marty for your reply.
I can access forum.intel.com , but if I want to add content 
or ask questions I need to sign-in and once I am logged in
there is no way to access forum, I keep being redirected as described above

MartyG-RealSense, 2019-06-26:
Once you have signed in and get redirected, please try getting back to the RealSense forum front page with this link:

https://forums.intel.com/s/topic/0TO0P00000018NTWAY/intel-realsense-technology

patrickpoirier51, 2019-06-26:
Well.... we are getting from bad to worse !!
Cannot sign in now,  the site is not responding .... HTTP ERROR 502

MartyG-RealSense, 2019-06-26:
With login problems on the Intel Support site, the Intel IT admins recommend trying clearing your browser cache.

patrickpoirier51, 2019-06-26:
Well, finally I could post here:
https://support.intelrealsense.com/hc/en-us/community/posts/360033190853-Autonomous-Flying-using-the-RealSense-T265-Tracking-Camera-in-GPS-Denied-Environment

Feel free to reference on the "old" forum, as we cannot get access ;-)

MartyG-RealSense, 2019-06-26:
Thanks so much for posting about Thien's excellent drone work!

patrickpoirier51, 2019-06-28:
Still pending approval, doesn't looks like an active and opened "community"  to me ... anyway, thanks for your help

MartyG-RealSense, 2019-06-28:
I recently had a comment stuck in the approval filter for 12 days before I noticed, so the person I was responding to wouldn't have seen it.  I would recommend stripping out any links you had in the message and posting it again to see if that gets you past the filter.

closed 2019-07-11 by patrickpoirier51

delebash, 2020-02-28:
patrickpoirier51

Did you do anything to resolve the redirect to sfederation issue.  I have the same problem, tried different browsers and even my phone browser but I keep getting redirected

MartyG-RealSense, 2020-02-28:
You should be able to access the forum by visiting this link directly.

https://forums.intel.com/s

You can then sign in at the top corner if you are not already signed in, and you should then be logged in and able to access your profile by left-clicking on your name and selecting the 'My Profile' option from the menu that drops down.

The old RealSense forum's front page that lists posts is no longer accessible as of January 2020 but you can search the forum for RealSense related posts in the forum's search box by including the word 'realsense' in your search keywords.

delebash, 2020-02-28:
Thanks, just contacted intel and they are aware of the login problem and working on it.  Ty

https://forums.intel.com/s<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fforums.intel.com%2Fs&data=02%7C01%7C%7C1c6ec71e72484e65747908d7bc781070%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637185096158260075&sdata=pCAU4eyBU2Gr6CRlHOXYy4coOu4jgDpplEjYZKuDqxM%3D&reserved=0>