realsenseai/librealsense #4295: Unable to use D435 on ROS with Jetson TX2

jerryt3204, 2019-06-26:
Did anyone try to use D435 on ROS with Jetson TX2 (Jetpack3.3) ?
I followed Jetsonhack's steps:
https://github.com/jetsonhacks/installRealSense2ROSTX
but there is an error:
E: Unable to locate package ros-kinetic-librealsense2

Before i did the step above, i've done these steps:
Installed librealsense2, reference:
https://github.com/jetsonhacks/buildLibrealsense2TX
Installed ROS-kinetic, reference:
https://github.com/jetsonhacks/installROSTX2

| Camera Model                       |  D435 | 
| Firmware Version                   | 05.11.04.00  | 
| Operating System & Version |  Ubuntu  16.04 | 
| Kernel Version (Linux Only)    |  4.14.38-tegra  | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            |  2.13.0   | 

Is there any solution to this problem ?
Thanks a lot.

eMrazSVK, 2019-07-02:
`git clone https://github.com/IntelRealSense/realsense-ros` into your catkin_ws/src directory and run catkin build (catkin_make). JetsonHacks is pretty good, but since it is one person project and librealsense is updating frequently, lot of things there are outdated. It should work.

jerryt3204, 2019-07-05:
@eMrazSVK Thanks for your reply!!
I've done some steps like yours, and it works well.
I will test it again if i flash ma TX2, thank you.

closed 2019-07-18 by RealSenseCustomerSupport