realsenseai/librealsense #4296: How to use hole filling  filters

buctfy, 2019-06-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.0                           | 
| Language                            |  C++/opencv                         | 

### Issue Description

I want to get the information of object 's distance all the time,but there are many black holes in my depth image.I used hole filling  filters(accoring to  https://dev.intelrealsense.com/docs/post-processing-filters) 
[image]
,there are no difference between two pictures.
[image]

Here are my codes:`using namespace std;
using namespace cv;
int main()
{
	rs2::colorizer c;
	rs2::pipeline pipe;
	rs2::config cfg;
	rs2::hole_filling_filter  hole_filter;
	hole_filter.set_option(RS2_OPTION_HOLES_FILL,1);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	rs2::pipeline_profile profile = pipe.start(cfg);
	
	
	while(1)
	{
		rs2::frameset frames = pipe.wait_for_frames();
		rs2::frame depth_frame = frames.get_depth_frame().apply_filter(c);
		rs2::frame filtered = depth_frame;

		
		Mat depth(Size(640,480), CV_8UC3, (void*)depth_frame.get_data(), Mat::AUTO_STEP);
		
		filtered = hole_filter.process(filtered);
		Mat filtereddepth(Size(640, 480), CV_8UC3, (void*)filtered.get_data(), Mat::AUTO_STEP);
		imshow("filteredDepth", filtereddepth);
		imshow("depth", depth);
		
		waitKey(10);
			
	}
	return 0;
}
`

menandro2015git, 2019-06-26:
I think you shouldn't apply the colorizer before you do the filtering.

AntoineWefit, 2019-06-27:
@buctfy you should take a look at the [rs-measure example](https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp), it might help you.

ev-mp, 2019-07-01:
@buctfy hello, as already noted by @menandro2015git - the filters apply to depth frames only.
The result of `rs2::frame depth_frame = frames.get_depth_frame().apply_filter(c);` is a RGB frame with 16bit depth data mapped to RGB using JET color scheme with histogram equalization. Hence calling `hole-filling` has not effect.
Make sure to apply the `colorizer` after the other (e.g. `hole-filling`) filters.

buctfy, 2019-07-04:
> 16bit depth data mapped to RGB using JET color scheme with histogram equalization. Hence calling `hole-filling` has not effect.

I delete some codes about colorizer,but when i run the code,depth image window don't show the image,and close after a little time
`using namespace std;
using namespace cv;
int main()
{
	//rs2::colorizer c;
	rs2::pipeline pipe;
	rs2::config cfg;
	rs2::hole_filling_filter  hole_filter;
	hole_filter.set_option(RS2_OPTION_HOLES_FILL,1);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	pipe.start(cfg);
	
	
	while(1)
	{
		rs2::frameset frames = pipe.wait_for_frames();
		rs2::frame depth_frame = frames.get_depth_frame();
		rs2::frame filtered = depth_frame;
		Mat depth(Size(640,480), CV_8UC3, (void*)depth_frame.get_data(), Mat::AUTO_STEP);
		
		filtered = hole_filter.process(filtered);
		Mat filtereddepth(Size(640, 480), CV_8UC3, (void*)filtered.get_data(), Mat::AUTO_STEP);
		imshow("depth", depth);
		imshow("filteredDepth", filtereddepth);
		
		
		waitKey(10);
			
	}
	`
[image]

menandro2015git, 2019-07-04:
Data type should be CV_16U or CV_16UC1.

closed 2019-07-22 by dorodnic