realsenseai/librealsense #4297: MulticameraIMU data mix up

menandro2015git, 2019-06-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.21                          | 
| Language                            |  C++                         | 

### Issue Description
I have two D435i. When both are plugged in, polling the IMU data appears to get mixed up. It happens even when the IMU stream is disabled on one camera, and I only read the IMU in another. The reading appears to be coming from both devices at random.

This can be replicated from the rs-multicam.cpp example by enabling only the gyro and accel.

menandro2015git, 2019-07-22:
I'm not sure why this isn't being addressed. I bought two D435i and the other one is useless because of this problem. BUMP

DSDev-NickHogle, 2019-10-15:
I too am having this issue.  We have a complex multi-camera setup with 5 cameras, and the IMU data is coming from all cameras at once.  We are trying to use the accelerometer data to estimate camera pose, but with no way to know which data comes from which camera, the data is not useful.

DSDev-NickHogle, 2019-10-29:
Following up here.   I tried upgrading firmwares on all the cameras, and updated the Windows UWP Driver to the latest available. Here is a description of my setup:

Property | Value
----------|-------
Camera Model | D435i
Windows Version | 10.0.18362 Build 18362
Firmware Version | 05.11.15.00
Platform | PC
SDK Version | 2.22.0.856
Language | C++, Python
USB Version | 3.2

dorodnic, 2020-03-04:
This should be resolved with the latest release

closed 2020-03-04 by dorodnic