realsenseai/librealsense #4300: T265 Localization map in Matlab

JimmyFroost, 2019-06-26:
| Camera Model | D415 + T265 |
| Firmware Version | (05.11.01.100) / (0.0.18.5502) |
| OS | Win ( 10) |
| Platform | PC |
| SDK Version | { 2.2} |
| Language |Matlab 2019a |
------------

Hi ,

I've been reading issues about saving localization maps, and testing it a bit in RealSense Viewer 2.21.0 :

[image]

I think I understand that it's possible with C++ and Python coding, is it already( I don't think so) or will it be (I strongly hope so) possible within Matlab  ?

RealSenseCustomerSupport, 2019-07-16:
----------------------------------------------

Current librealsense not support this feature yet. will check its feasibility

JimmyFroost, 2019-07-16:
Thank you, I'm looking forward to it.

Good luck

closed 2020-02-03 by RealSenseCustomerSupport