realsenseai/librealsense #4305: Horizontal object distance problem.

vusdo, 2019-06-27:
The distance between the straight line and the oblique line must be very different between the camera and the object.

However, checking with the depth.get_distance function is not so different.

The object color on the right side of the screen should not be blue.

I want to know why.

[image]

dorodnic, 2019-06-27:
Could you please share IR / RGB images as well?

vusdo, 2019-06-27:
In RGB, the object is not displayed on the screen because the angle of view is small.

dorodnic, 2019-06-27:
I'm not sure I understand. Is the color-scale on the right not accurately depicts object distances? Or are the distances from get_distance wrong?

vusdo, 2019-06-27:
The actual measurement result is that the middle object is the closest, and the object should be farther away on the oblique side.  However, get_distance results in very similar values.

vusdo, 2019-07-03:
Do you understand? I want to know the reason.

agrunnet, 2019-07-03:
I don’t quite understand what you mean by oblique angle. You mean the objects in left and right side should be farther away?
When you consider distance to camera for your ground truth measurement do you measure from camera or from the plane of the camera? The depth map measures the distance in Z axis. If you want actual total distance directly away from the camera point you need to convert to pointcloud and then measure distances. 
I am not sure if this helps.

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@vusdo Do you still have questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@vusdo Didn't get response from you for weeks. If no other questions, will close it accordingly. Thanks!

closed 2019-10-12 by RealSenseCustomerSupport