realsenseai/librealsense #4306: How do I save all the aligned color and depth frames in a.bag file?

nevynwong, 2019-06-27:
| Camera Model                       |D415| 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| Language                            |MATLAB | 

I've recorded a.bag file and aligned the depth and color, but how do I save all the aligned frames?Because I need to align the color image sequence and the depth sequence or point cloud.
My alignment code looks like this.

pipe = realsense.pipeline;
config = realsense.config;
config.enable_device_from_file('11.bag');
profile = pipe.start(config);
align = realsense.align(realsense.stream.color);

frameset = pipe.wait_for_frames();
color_frame = frameset.get_color_frame();
frameset = align.process(frameset);
aligned_depth_frame = frameset.get_depth_frame();

I don't know if this is the right thing to do. Can you help me?I'm in a hurry.Thanks.

jb455, 2019-06-27:
If you save the frames normally, you can then align them when you play them back later.

nevynwong, 2019-06-27:
I don't quite understand this. Could you tell me how to do it specifically?Thanks.

jb455, 2019-06-27:
I don't know specifically how you do it using the matlab wrapper, but the basic process would be: save the colour and depth frames in the bag file (no need to do alignment at this stage unless you need it for something else while recording), then when playing back align the frames as required. The code you've posted looks reasonable for this purpose (the second part), unless matlab needs anything special. If you're getting any errors etc with this code, post them to help others understand what's going wrong.

nevynwong, 2019-06-27:
The above code aligns depth to color from .bag file,  but I don't know how to extract the aligned color and depth sequence. Coulc the aligned color and depth frames be saved in a.bag and then extract them?But that doesn't seem right.

dorodnic, 2019-06-27:
Bag file records raw data from the camera. Using this data, you can perform alignment when you need aligned frames back.

nevynwong, 2019-06-28:
Yes, I now need to align the data recorded in the.bag file, but how to extract the aligned color and depth sequences is difficult for me.

menandro2015git, 2019-06-28:
Your code seems correct. To save aligned_depth_frame, you need to extract the data. From depth_example.m:

`depth= aligned_depth_frame.get_data();`

nevynwong, 2019-06-28:
It's just one frame. What I want to save is the whole video sequence images.

nevynwong, 2019-07-01:
@dorodnic, I now need to align the depth and color in the collected. Bag and then do something else.But it looks like I can't extract all the aligned frame sequences with the above processing, so what should I do?

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@nevynwong Sorry for late response. Which SDK and FW version did you use when get the issue? Did you still have the issue with the latest SDK and FW version? Looking forward to your update. Thanks!

kjacks21, 2019-09-25:
@RealSenseCustomerSupport to avoid opening another issue, and since @nevynwong said they wanted the whole video sequence and the code below will probably help them, I'll write to here. When I align the color and depth frames and try to access both, I always get the following error as a result of running the code below:

**Error**
```
depth_image = np.asanyarray(aligned_depth_frame.get_data())
RuntimeError: null pointer passed for argument "frame_ref"

```

**Code**
```
import argparse
import pyrealsense2 as rs
import numpy as np
import cv2
import os

def bag2mp4(bag_fname, color_mp4_fname, depth_mp4_fname):

    config = rs.config()
    pipeline = rs.pipeline()

    # make it so the stream does not continue looping
    rs.config.enable_device_from_file(config, bag_fname, repeat_playback=False)
    config.enable_stream(rs.stream.color)
    profile = pipeline.start(config)
    # this makes it so no frames are dropped while writing video
    playback = profile.get_device().as_playback()
    playback.set_real_time(False)

    colorizer = rs.colorizer()

    align = rs.align(rs.stream.color)

    i = 0
    while True:

        # when stream is finished, RuntimeError is raised, hence this
        # exception block to capture this
        try:
            frameset = pipeline.wait_for_frames(timeout_ms=100)
        except (RuntimeError):
            print('frame count', i-1)
            break

        frameset = align.process(frameset)

        aligned_color_frame = frameset.get_color_frame()
        color_image = np.asanyarray(aligned_color_frame.get_data())

        # convert color image to BGR for OpenCV
        r, g, b = cv2.split(color_image)
        color_image = cv2.merge((b, g, r))

        # Update color and depth frames:
        aligned_depth_frame = frameset.get_depth_frame()
        depth_image = np.asanyarray(aligned_depth_frame.get_data())

        break

return None
```

The example [here](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb) only has one frame and other issues about extracting different types of frames using conditionals didn't resolve my issue.

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@nevynwong Could you please update if you still have the issue with the latest SDK and FW?
@kjacks21 Please feel free to create another new issue for your case. Thanks!

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@nevynwong We didn't get response from you for weeks. Will close this at this point. Please feel free to create another one if you still have issues or questions. Thanks!

closed 2020-03-31 by RealSenseCustomerSupport