realsenseai/librealsense #4307: get_fisheye_frame(1).as_video_frame()  fuction don't give any output in T265

malith1992, 2019-06-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |   Ubuntu 18 .04                                      |            
| Platform                                 | Jetson nano                                            |
| Language                               |  python 2.7                                             | 
| Segment			         |  Robot                                                    | 

I tried to run this file [t265_stereo.py ](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py)

But the code is not working because of an issue in the following code segment. (Line 94 - 95)

frameset = frame.as_frameset() 
f1 = frameset.get_fisheye_frame(1).as_video_frame() 

Why is the f1 empty when I run the code

RealSenseCustomerSupport, 2019-07-17:
----------------------------------------------

Hi @malith1992,

You can also have a try of below example to see how it works.
https://github.com/IntelRealSense/librealsense/pull/4442/files

RealSenseCustomerSupport, 2019-07-31:
----------------------------------------------

Hi @malith1992,

any update? Is the issue resolved?

malith1992, 2019-08-07:
> Hi @malith1992,
> 
> You can also have a try of below example to see how it works.
> https://github.com/IntelRealSense/librealsense/pull/4442/files

It works well. Thank you very much

closed 2019-08-07 by malith1992