realsenseai/librealsense #4308: Require T265 calibration for underwater use

ronzeman99, 2019-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (v2.23.0) | 
| Operating System & Version |   {Linux (Ubuntu 16)| 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  {}                          | 
| Language                            |  {C}                          | 
| Segment			|  {Robot}                          | 

As above, I need to be able to set lens distortion parameters of this camera in order to use it underwater. After reading through the documentation it doesn't appear that it is possible to set this for the T265 camera, is this true?

Is there any plan for adding this capability in the future? Is there a work around that I can use in the meantime?

Appreciate any help.

PaulKrush, 2019-06-28:
Wow. Are the intrinsics different underwater? I would believe this. Have you done this and have calibration images to share?

ronzeman99, 2019-07-02:
I have noticed that the init_stream_profiles() function in tm2/tm-device.cpp has a set_intrinsics function. Will this allow the functionality I am looking for @dorodnic?

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

Hi @ronzeman99 ,

Are you going to use your own method to calibrate the intrinsics and replace our calibration data in your use case?
You prefer to have the custom calibration capability like D400 camera?

ronzeman99, 2019-07-09:
Hi @RealSenseCustomerSupport,

In my use case I would calibrate the intrinsics myself and then replace your calibration data. 
It would also be very useful to have the support for custom calibration like the D400 has.

Cheers,
Ron

ronzeman99, 2019-07-12:
Would it be possible to provide an indication of whether this is possible given the current SDK and whether it may be on the road map for future releases? We are in the early stages of module selection for a consumer product and this function is a requirement for us. @dorodnic @RealSenseCustomerSupport 

Appreciate your time.

dorodnic, 2019-07-22:
Hi @ronzeman99 
Sorry for the delay. It is not possible in current SDK. AFAIK this capability will be available to module (bulk) customers, and may be also provided for the T265 dev-kit (cannot commit on that). @bfulkers-i - do you have any information on this?

bfulkers-i, 2019-07-22:
In the next few months we will add librealsense support for setting custom intrinsics, but the intrinsics will have to follow the distortion model used by T265. T265 uses the Kanalla-Brandt distortion model which is implemented in [OpenCV as the "fisheye" module](https://docs.opencv.org/3.4/db/d58/group__calib3d__fisheye.html).

If you have already tested the T265 underwater, we would love to see recordings of that @ronzeman99.

CC @dorodnic

dorodnic, 2019-10-15:
@ronzeman99 - APIs for writing custom calibration have been added to [SDK 2.27](https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2270). Please take a look and let us know if this can help. I'm not sure if existing distortion model would work for your use-case, but it might be good enough.

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @@ronzeman99,

Do you have any update to utilize the APIs for your own calibration?

closed 2019-11-11 by dorodnic

1E0-may, 2022-01-02:
@RealSenseCustomerSupport 
Hello, I also need to update the camera parameters for underwater usage.
I checked the C++ API, and I'm unsure how to use it... is there any example or tutorial?
thanks