realsenseai/librealsense #4316: On Jetson TX2: No RealSense devices were found! Use Realsense t265

kehanXue, 2019-06-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } |
| Firmware Version                   | 0.0.18.5715 |
| Operating System & Version |  Linux (Ubuntu 16) |
| Kernel Version (Linux Only)    |  (4.4.38)                                         |
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  { 2.23.0 }                          |
| Language                            |  { Cpp }                          |
| Segment			|  {Robot }                          |

### Issue Description
When I connect my realsense t265 to my Jetson TX2, and launch the ros node, the output shows blow:
```shell
nvidia@tegra-ubuntu:~/librealsense$ roslaunch realsense2_camera rs_t265.launch
... logging to /home/nvidia/.ros/log/dad7f92e-98dd-11e9-bce6-00044ba51658/roslaunch-tegra-ubuntu-3854.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:39786/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 62
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 800
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 848
 * /camera/realsense2_camera/gyro_fps: 200
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/topic_odom_in: camera/odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [3864]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dad7f92e-98dd-11e9-bce6-00044ba51658
process[rosout-1]: started with pid [3877]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [3894]
process[camera/realsense2_camera-3]: started with pid [3895]
[ INFO] [1561641426.044402171]: Initializing nodelet with 6 worker threads.
[ INFO] [1561641426.307043678]: RealSense ROS v2.2.6
[ INFO] [1561641426.307168516]: Running with LibRealSense v2.23.0
[ WARN] [1561641426.325352177]: No RealSense devices were found!
[ WARN] [1561641432.326449590]: No RealSense devices were found!
[ WARN] [1561641438.327410692]: No RealSense devices were found!
```
I make sure that I have connected the realsense t265 device courrectly.

I have tried to rebuild the TX2's kernel. And I have tried two ways.
1. First I have tried to use the [Jetsonhack's scripts](https://github.com/jetsonhacks/buildLibrealsense2TX) to rebuild the TX2's kernel. And I modified the version of librealsense variable in the scripts to **v2.23.0**. All the actions were successful. But when I reboot the TX2, it still didn't work.
2. Then I tried to rebuild the kernel by myself with the scripts [here](https://github.com/jetsonhacks/buildJetsonTX2Kernel), which also supported by Jetsonhack. I changed the options in the xconfig window by referring the options in Jetsonhack's buildLibrealsense2TX respo. Most of them are about **IIO and HID**. Then I rebuild the kernel successfully and reboot, and then rebuild the librealsense. But the ROS nodelet doesn't work, too. The message "**No RealSense devices were found!**" was always showing.

The cmake version is 3.5.1, and the Jetpack version I used is 3.3, matched with L4T R28.2.1.
```shell
nvidia@tegra-ubuntu:~$ uname -r
4.4.38-tegra-realtimes
```
The *realtimes* means the carrier board's brand which TX2 used. My version is [RTSO-9003U](http://www.realtimes.cn/cn/product/rtso-9003U.html). **To adapted with the carrier board, I have update the kernel one time before. I think maybe this cause the problem?** :(

When I lsusb, the outpus shows below:
```shell
nvidia@tegra-ubuntu:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 03e7:2150  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

And when I build the librealsense, because the Internet connection is unstable, I didn't build the librealsense with TM2. Does this may cause the problem?

I have worked on this problem four days, I am really hopeless. Is the any step I have forgot?
I will be very appreciate  if you can help me slove my problem! Thanks a lot!

kehanXue, 2019-06-27:
Oh I have just see the #4285. Let me have a try building with TM2.

kehanXue, 2019-06-28:
Oh I have fixed it. Just building with TM2 and the t265 will work on Jetson tx2 well!

closed 2019-06-28 by kehanXue

ArtlyStyles, 2019-07-08:
@kehanXue 

Can you share how you installed all librealsense on TX2? I have problem to install the software package.

kehanXue, 2019-07-24:
> @kehanXue
> 
> Can you share how you installed all librealsense on TX2? I have problem to install the software package.

Just git clone the librealsense, and build with TM2. Then create the udev rules for t265 with the script in librealsense script dir. Finally reboot the tx2, it will be ok. What are the specific details of the problem you encountered?

jingshaojing, 2020-10-01:
> Oh I have fixed it. Just building with TM2 and the t265 will work on Jetson tx2 well!

sorry,can you help explain what's mean with building with TM2 ?what is TM2?

jingshaojing, 2020-10-01:
> > Oh I have fixed it. Just building with TM2 and the t265 will work on Jetson tx2 well!
> 
> sorry,can you help explain what's mean with building with TM2 ?what is TM2?

sorry,I have know how to do .thanks