realsenseai/librealsense #4319: Unable to use 60 fps, Realsense D435

atyshka, 2019-06-27:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.18.0-24-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.23.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                      | 

Running the latest firmware and sdk and I cannot get the camera to reliably stream color at 60 fps at any resolution. I know that I can't stream full res at 60fps, but even low resolutions don't work. I initially noticed the issue using the ROS wrapper, but I also have the problem in the realsense viewer. It is using a USB 3.0 connection. Sometimes I can get 60 fps color but then it drops down to 30fps. Depth streams just fine at 60fps.

dorodnic, 2019-06-27:
Maybe it is due to auto-exposure? There is an option call auto-exposure priority-mode that will prevent FPS reduction in low light settings.

atyshka, 2019-06-27:
That seems to solve my problem... but while trying to solve it myself i found another problem: the uvc module isn't correct. I installed librealsense2-dkms, rebooted, but running `modinfo uvcvideo | grep version` prints out `version:        1.1.1` which isn't correct. Any ideas why this is happening?

atyshka, 2019-07-01:
Closing this since the original issue is solved. but please see #4343 regarding the above.

closed 2019-07-01 by atyshka