realsenseai/librealsense #4320: Data hole on surface reflecting light

m1halo, 2019-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  5.11.6.250 | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC  |
| SDK Version                            |  n/a                          | 
| Language                            |  Viewer                          | 
| Segment			|  others                          | 

### Issue Description
We run into issue with data holes. I will do my best to describe the conditions:

Certain surfaces reflecting light on angle are causing depth data to have data holes. In the example below left hand (right on a photo) has oily cream applied on it. Right hand has no product. Light source (ie. halogen lamp) is directed on subject (behind camera).

Is this a known limitation or would you suggest any setting for the camera that could improve scan in this case?

[image]
[image]
[image]

MartyG-RealSense, 2019-06-28:
An old trick for dealing with gaps in data caused by reflectivity is to use a fine spray-on powder such as foot powder or baby powder to dampen the reflection.  It is reputed that a project to scan a vehicle bodywork repair workshop sprayed the entire workshop in baby powder to remove the reflections of the metalwork.

There are also professional 3D scanning sprays that can do the same job of dampening reflections.   An example use is on jewelry that has to be scanned for a catalog.

https://www.laserdesign.com/3d-scan-spray/?language=en_US

Another way to dampen reflection on large areas is to apply an anti-reflection film to it.  google for 'anti reflection film' for more details.

dorodnic, 2019-06-28:
In the image above I don't think the issue is reflectivity (strictly there is no reflection in the IR image), but rather IR over-saturation. You can try experimenting with different values of exposure and gain controls to fine tune the image to your use case. 
In extreme case where you have brightly illuminated surfaces and really dark surfaces, it is impossible to find one value of sensor exposure without having either completely black or completely white regions, but for this case, the camera does offer a control to rapidly switch between two values of exposure.

closed 2019-07-22 by dorodnic