realsenseai/librealsense #4322: Different D435i IMU frame in different SDK version

paulyang1990, 2019-06-28:
I have two ubuntu computer, one is 16.04 and the other is 18.04.

I used the same steps to setup realsesnse lib, ros wrapper and launch file. But I astonishingly found that the two computer gave IMU data in different frames. 

On the 16.04 computer, IMU frame is aligned with camera frame. While on the 18.04 computer, IMU frame is different. 

I guess the reason is that in SDK, code to read IMU data and process its frame on 16.04 and 18.04 are different. Please confirm this problem. Thanks.

dorodnic, 2019-06-28:
No such mechanism in the SDK. Please check what versions of the SDK and the ROS wrapper are being used across the two devices.

ev-mp, 2019-07-01:
@paulyang1990 hello, please provide logs/screenshots with the different orientations reports

ev-mp, 2019-07-10:
@paulyang1990 , any update ?

paulyang1990, 2019-07-15:
@ev-mp This problem disappears after I update ROS wrapper. I cannot reproduce the problem. But I am sure that I did observe this problem with some SDK and ROS wrapper combinations.

ev-mp, 2019-08-05:
Closing for now - please reopen if you manage to reproduce it

closed 2019-08-05 by ev-mp