realsenseai/librealsense #4323: How to display the fish eye camera view of T265

malith1992, 2019-06-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |  Ubuntu 18.04                                        | 
| Platform                                 | NVIDIA Jetson nano  |
| SDK Version                            | build from source                     | 
| Language                            |  python }                          | 
| Segment			|   Robot             | 

I installed pyrealsense2 and it works perfectly without giving any errors. But I can't find a function in pyrealsense2 library to display the camera view of T265. Can someone give me a solution please ?

eMrazSVK, 2019-07-04:
Hey,

Don't know much about it, but I think you can get it to work, if you go through this [example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py). Exactly line 140, where you see, that how you can get fisheye streams.

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

Hi @malith1992,

Is the issue resolved based on @eMrazSVK suggestion?

malith1992, 2019-07-11:
@RealSenseCustomerSupport @eMrazSVK  This code [example](https://raw.githubusercontent.com/IntelRealSense/librealsense/master/wrappers/python/examples/t265_stereo.py) is not running properly

RealSenseCustomerSupport, 2019-07-24:
----------------------------------------------

Hi @malith1992,

1. For 't265_stereo.py" example, you can try to comment below code and see how it works.

    _#if key == ord('q') or cv2.getWindowProperty(WINDOW_TITLE, cv2.WND_PROP_VISIBLE) < 1:_
    _# print('No cv2 window')_
    _# break_

2. How about below example?
https://github.com/IntelRealSense/librealsense/pull/4442/files

RealSenseCustomerSupport, 2019-07-31:
----------------------------------------------

Hi @malith1992,

Is the issue resolved. If no additional support required, will close this.

closed 2019-08-06 by RealSenseCustomerSupport