realsenseai/librealsense #4324: Creating a 3D point cloud in python

a17sans, 2019-06-28:
Hello everyone, 

Is there a way to create a .PLY image only with python (without using RealSense Viewer) using only a camera and a rotating plateform ? 

I would like to place an object on the plateform, film all the rotation and get a .ply image all with python code. Does anyone know if it is possible ? 

Thanks a lot,

Alric

bill-wright, 2019-06-28:
Hi Alric,

This example should get you started in the right direction:

[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py](url)

a17sans, 2019-06-28:
Brilliant thank you so much

a17sans, 2019-07-01:
I used this example, but I do not manage to record during a certain period of time. 
I think this is what explains that I only get a few points on the .PLY and no object form.

Is there a way to record a video and generate a .PLY after that ? 

Thank you for your help

a17sans, 2019-07-05:
Created other issue with more description  and code here : [#4370](https://github.com/IntelRealSense/librealsense/issues/4370)

closed 2019-07-05 by a17sans