realsenseai/librealsense #4325: What is the correct way to shut down?

vhanded, 2019-06-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-52-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.23                          | 
| Language                            |  c++                         | 
| Segment			| Robot                          | 

### Issue Description
What is the correct way to shut down realsense? Any specific step to do before terminate?
The reason is because, every time I shut down, I need to unplug USB, plug in again to work, else `poll_for_frame()` will always return false.

eMrazSVK, 2019-07-04:
What do you mean by shut down exactly? If it is on API level, you need to call `pipe.stop()`. If it is on realsense-viewer level, switching the sensor in the app should be sufficient.

Samvid95, 2019-07-17:
So in C code you can delete the `cfg` object and `release frames` as well. But do we have that in C++? Or just `pipe.stop()` is just enough?

RealSenseCustomerSupport, 2019-08-08:
----------------------------------------------

There is also an article on the GitHub to describe the SDK's frame management:
https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0
Closing this ticket if no further question. Thank you.

closed 2019-08-15 by RealSenseCustomerSupport