realsenseai/librealsense #4326: Errors compiling nodejs wrappers

BenBergman, 2019-06-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    |  macOS v10.14.5                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.23.0 (cloned from git master)                          | 
| Language                            |  nodejs                          | 
| Segment			|  others                        | 

### Issue Description
Trying to build nodejs examples fails with the following error after running `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_NODEJS_BINDINGS=1 && make`

```
...
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/v8.h:2561:3: note: 'BooleanValue' has been explicitly marked deprecated here
  V8_DEPRECATED("BooleanValue can never throw. Use Isolate version.",
  ^
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/v8config.h:307:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated(message)))
                            ^
In file included from ../src/addon.cpp:9:
In file included from ../node_modules/nan/nan.h:203:
In file included from ../node_modules/nan/nan_new.h:189:
../node_modules/nan/nan_implementation_12_inl.h:103:42: error: no viable conversion from 'v8::Isolate *' to 'Local<v8::Context>'
  return scope.Escape(v8::Function::New( isolate
                                         ^~~~~~~
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/v8.h:183:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'v8::Isolate *' to 'const v8::Local<v8::Context> &' for 1st argument
class Local {
      ^
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/v8.h:183:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion from 'v8::Isolate *' to 'v8::Local<v8::Context> &&' for 1st argument
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/v8.h:187:13: note: candidate template ignored: could not match 'Local<type-parameter-0-0>' against 'v8::Isolate *'
  V8_INLINE Local(Local<S> that)
            ^
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/v8.h:4126:22: note: passing argument to parameter 'context' here
      Local<Context> context, FunctionCallback callback,
                     ^
In file included from ../src/addon.cpp:9:
In file included from ../node_modules/nan/nan.h:203:
In file included from ../node_modules/nan/nan_new.h:189:
../node_modules/nan/nan_implementation_12_inl.h:337:37: error: too few arguments to function call, expected 2, have 1
  return v8::StringObject::New(value).As<v8::StringObject>();
         ~~~~~~~~~~~~~~~~~~~~~      ^
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/v8.h:5380:3: note: 'New' declared here
  static Local<Value> New(Isolate* isolate, Local<String> value);
  ^
In file included from ../src/addon.cpp:9:
In file included from ../node_modules/nan/nan.h:203:
In file included from ../node_modules/nan/nan_new.h:189:
../node_modules/nan/nan_implementation_12_inl.h:337:58: error: expected '(' for function-style cast or type construction
  return v8::StringObject::New(value).As<v8::StringObject>();
                                         ~~~~~~~~~~~~~~~~^
../node_modules/nan/nan_implementation_12_inl.h:337:60: error: expected expression
  return v8::StringObject::New(value).As<v8::StringObject>();
                                                           ^
In file included from ../src/addon.cpp:9:
../node_modules/nan/nan.h:839:18: warning: 'MakeCallback' is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
    return node::MakeCallback(
                 ^
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/node.h:173:1: note: 'MakeCallback' has been explicitly marked deprecated here
NODE_DEPRECATED("Use MakeCallback(..., async_context)",
^
/Users/benjamin.bergman/.node-gyp/12.3.1/include/node/node.h:93:20: note: expanded from macro 'NODE_DEPRECATED'
    __attribute__((deprecated(message))) declarator
...
```

There are several other errors, but this is the first few.

alexeden, 2019-08-01:
@BenBergman I just tried installing the package for the first time and it looks like the Node wrapper is using nan to bind to C++, which has been deprecated in favor of [N-API](https://nodejs.org/api/n-api.html) because nan is tightly coupled to the version of node you're using.

```
$  nvm use 10
```
^ This is a workaround, but really the wrapper library needs to be rewritten using N-API.

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

@alexeden Thanks very much for your input!
@BenBergman Did you get chance to try what Alexeden suggested workaround? Looking forward to your feedback. Thanks!

RealSenseCustomerSupport, 2019-09-02:
----------------------------------------------

@BenBergman Any update? Thanks!

BenBergman, 2019-09-07:
I haven't had a chance to try it. I ended up building something in OpenFrameworks instead while I waited for a solution here and haven't had the time to try this out yet.

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@BenBergman Did you get chance to try later? Looking forward to your update. Thanks!

Binklebonk, 2019-11-25:
@RealSenseCustomerSupport , in a recent project (this month) I tried to use the Node.js wrapper and spent days wrestling with the development environment and trying to get it to build.  I even formatted a fresh system with the most ideal versions of everything, in case there was some sort of conflict, with no success.

In the end, I used Python, which was far easier to get working.

closed 2019-11-26 by RealSenseCustomerSupport

Alex-OPTIM, 2020-05-05:
Suggestion by @alexeden worked well for me! I'm using MacOS with `nvm use 10`