realsenseai/librealsense #4328: Driver not recognized Windows 7

amashal, 2019-06-28:
I am not able to install the Intel 435i on Windows 7. I installed the latest SDK (2.23.0.909) multiple times but I have not been successful. Any suggestions?

Screenshot of my Device Manager:

[image]

| | |
|---|---|
|**librealsense**|2.23.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

MartyG-RealSense, 2019-06-29:
Someone else who had a driver detection problem when trying to install the camera on Windows 7 was given the advice in the link below.

https://github.com/IntelRealSense/librealsense/issues/2506#issuecomment-428653246

amashal, 2019-06-30:
I did try that, but unfortunately it did not work.

jkarvounis, 2019-07-01:
I have the same issue.  Tried updating USB 3.0 drivers did not help either.  Currently stuck since my computer cannot recognize my 435i

sherrysun2018, 2019-07-02:
Please note that D435i currently not supported on Win7 yet. You can find this in Release Note.
"D435i requires FW version 5.11.2.+ and is currently supported on Linux and Windows 10 only. Windows 7, Android and Mac OS support under development to be included in future releases."

closed 2019-07-16 by RealSenseCustomerSupport