realsenseai/librealsense #4331: Issue with t265 usb device identification

orshefi, 2019-06-30:
Hi guys,

I had looked further into the issue of Realsense T265 device sometimes need to be reconnected in order to be identified by the Realsense SDK.

After I sniffed around the code base I believed I arrived at the source of the issue:

I added the following print line in order to evaluate the reason for why the device isn't initialized despite it is connected:
`bool perc::UsbPlugListener::identifyDevice(libusb_device_descriptor* desc)
{
    std::cout << std::hex << desc->idVendor << ":" << std::hex << desc->idProduct << " usb type: " << desc->bcdUSB << std::endl;
    if ((desc->idProduct == TM2_T260_PID || desc->idProduct == TM2_T265_PID) && (desc->idVendor == TM2_DEV_VID) && (desc->bcdUSB >= USB_2_0))
    {
        return true;
    }

    return false;
}
`
The printed output was the following:
1d6b:3 usb type: 310
1d6b:2 usb type: 200
bda:411 usb type: 300
1d6b:3 usb type: 300
**3e7:2150 usb type: 110
3e7:2150 usb type: 110**
bda:5411 usb type: 200
46d:c534 usb type: 200

As you can see the device identifies as usb 1.1 stright from libusb interface, I'm still investigating for why does it happen (kernel or device firmware).

Share your thoughts about it, maybe you'll set me up for a quick solution

orshefi, 2019-07-01:
@ev-mp please share your thoughts

radfordi, 2019-07-02:
Hi @orshefi,

Thank for your detailed report.   I believe this is the sign of a T265 power on race condition that we have seen with certain hubs.

Are you using a hub, would you try the suggestion in #3657 to power the hub first and then plug in the T265?  If not could you let us know what your configuration is?

orshefi, 2019-07-03:
Hi @radfordi,

From what I could understand from the information provided in your answer and the ticket attached its a firmware issue.

The solution offered on #3657 isn't applicable to most purposes, one of the main abilities of the t265 camera is that it has lean bandwidth that allows it to be trafficked by a USB hub together with other cameras.

Most of the robotics application use embedded ARM arch computers which don't have a lot of USB port and use USB hubs in order to connect multiple Realsense devices.

In order to get a device that uses Realsense T265 to be commercial grade there isn't a scenario in which the user will have to disconnect and connect the camera on every boot (the mere sound of it is ridiculous).

In my case the USB hub is part of my proprietary PCB (not external) f.e., do you suggest my users will boot my product when the t265 is disconnected and after boot connect it?

Please try to explain the issue better, are you planning on dealing with it?

radfordi, 2019-07-09:
@orshefi, above you are seeing the Myriad 2 boot loader (`3e7:2150`) which runs from ROM before we upload the T265 firmware.   While we investigate why certain hubs are having this issue, can you confirm if that unplug/re-plug work-around works for you?

orshefi, 2019-07-09:
Hi @radfordi, 
Repluging works.
You mentioned that it doesn't work with certain hubs, can you specify the ones that it does work with?

hangmann, 2019-07-09:
> Hi @radfordi,
> Repluging works.
> You mentioned that it doesn't work with certain hubs, can you specify the ones that it does work with?

Hi orshefi,
replugging works for me, too. But I have the same problem as stated above. Personally I am using the Icy Box USB 3.0 Hub IB-AC6110

RealSenseCustomerSupport, 2019-07-24:
----------------------------------------------

As @radfordi mentioned, the Myriad 2 boot loader (`3e7:2150`) which runs from ROM before we upload the T265 firmware. The major problem is due to the power-on race condition between Myriad 2 and the hub. Regarding the HW workaround, the host (hub side) must be ready to perform the enumeration process the moment the TM2 is “plugged in” (powered up). Any other scenarios are likely to result in the above issue.
That's the reason T265 device will get into a lower USB protocol state as USB 1.1.

You can also refer to ##4113 to see if it can help in this case.

RealSenseCustomerSupport, 2019-07-31:
----------------------------------------------

Hi @orshefi,

Any update?

RealSenseCustomerSupport, 2019-08-06:
----------------------------------------------

Hi @orshefi, @hangmann,

Is the issue clarified? Still need any additional support for this topic?

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

I will close this

closed 2019-10-17 by RealSenseCustomerSupport

radfordi, 2020-01-31:
@orshefi, @hangmann, [many boards/hubs](https://github.com/mvp/uhubctl#compatible-usb-hubs) have the ability to power cycle individual ports.   We have found that running this program https://github.com/mvp/uhubctl once on startup gets the device working when the Movidus bootloader (3e7:2150) its stuck in USB1.0 mode.

Geonhee-LEE, 2020-03-29:
@radfordi  Hi, I found the USB connection problem like above. Let me know how to use [uhubctl](https://github.com/mvp/uhubctl)?  In addition, Python file can work autonomously using uhubctl?