realsenseai/librealsense #4332: D435i: memory usage increasing constantly

terranas, 2019-07-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.19.0 | 
| Language                            | C++ | 

### Issue Description
https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md
Processing IMU sensor data according to the above, process memory (Private Bytes) has increase about 27KB/sec.
Leave it running 36 hours, memory usage keep increased and process has died.
Strangely, I got same result by more simple code.
```
#include <librealsense2/rs.hpp>

int main()
{
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_GYRO);
    cfg.enable_stream(RS2_STREAM_ACCEL);

    rs2::pipeline pipe;
    pipe.start(cfg);
    getchar();
}
```

This looks like a memory leak to me. Is there a suspected cause?

ev-mp, 2019-07-01:
@terranas hello, thank you for the report.
Seem like a possible bug. We'll look into this

JamesPiggott, 2019-07-24:
I am having a similar issue using the Intel D415.

Required Info |  
-- | --
Camera Model | D415
Firmware Version | 05.11.06.250
Operating System & Version | Ubuntu 18.04
Platform | PC
SDK Version | 2.24.0
Language | C++

Just so we are on the same page I used the code example provided by terranas and evaluated it with Valgrind.
[valgrind_report.txt](https://github.com/IntelRealSense/librealsense/files/3426000/valgrind_report.txt)

tixidi, 2019-07-30:
I've met the same problem. Have you got any solution?

hkooverjee, 2019-07-30:
Hi all, we have also experienced similar issues running the same setup @JamesPiggott used. We have also tested on Ubuntu 16.04 which gave similar results.

Also, the valgrind report @JamesPiggott attached is similar to the one in #3433 (GitHub cannot reference the issue correctly). 

Downgrading to SDK 2.17.1, a version we previously used, gives no memory leak.

mbechard, 2019-08-08:
This seems to be affecting us as well. Any updates? We are on 2.19.1 but are happy to update to one where it's fixed

RealSenseSupport, 2020-07-01:
@terranas This issue was fixed in the current SDK release. Could you please give it a try?

RealSenseSupport, 2020-07-07:
@terranas If we don't hear from you in 7 days, this issue will be closed.

closed 2020-07-14 by RealSenseCustomerSupport