realsenseai/librealsense #4333: Unable to set RS2_FORMAT_RGB8 to infrared.

vhanded, 2019-07-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.23                           | 
| Language                            |  C++ | 
| Segment			|  others                          | 

### Issue Description
I want to get RS2_FORMAT_RGB8 for my infrared 1, so I enable the stream like this:

`  configuration_->enable_stream(RS2_STREAM_INFRARED, 0, 1280, 720, RS2_FORMAT_RGB8, 30);`

This returned error: `Couldn't resolve requests`

However, setting the format to `RS2_FORMAT_Y8` works.

When I run `rs-enumerate-devices`, I can see that RS2_FORMAT_RGB8 with the supported stream, supported resolution, and supported fps.

```
Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1920x1080	@ 25Hz	   Y16
    Infrared 2	  1920x1080	@ 25Hz	   Y16
    Infrared 2	  1920x1080	@ 25Hz	   Y8
    Infrared 1	  1920x1080	@ 25Hz	   Y8
    Infrared 1	  1920x1080	@ 15Hz	   Y16
    Infrared 2	  1920x1080	@ 15Hz	   Y16
    Infrared 1	  1920x1080	@ 15Hz	   Y8
    Infrared 2	  1920x1080	@ 15Hz	   Y8
    Infrared	  1280x720	@ 30Hz	   UYVY
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared	  1280x720	@ 30Hz	   BGRA8
    Infrared	  1280x720	@ 30Hz	   RGBA8
    Infrared	  1280x720	@ 30Hz	   BGR8
    Infrared	  1280x720	@ 30Hz	   RGB8   <-----

```
Notice it has no index number after `Infrared`, so while enabling stream, i omitted the index parameter, like this:

`  configuration_->enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_RGB8, 30);`

This shows another error:
`Profile does not contain the requested stream`

In `realsense-viewer`, it is supported, which makes me believe, there must be some configuration that I missed, else this is a bug.

Any help?

menandro2015git, 2019-07-01:
The index should be 1 or 2, not 0.

vhanded, 2019-07-02:
> The index should be 1 or 2, not 0.

Thanks, edited 0 to 1. Issue still the same.

> @vhanded hello,
> The syntax seem correct, can you share the full `configuration_->enable_stream(..;` segment for all streams requested ? Attaching the full log of `rs-enumerate-devices -o` will be useful as well.

Here i the full log:
```
    stream       resolution      fps       format   
    Infrared 1	  1920x1080	@ 25Hz	   Y16
    Infrared 2	  1920x1080	@ 25Hz	   Y16
    Infrared 2	  1920x1080	@ 25Hz	   Y8
    Infrared 1	  1920x1080	@ 25Hz	   Y8
    Infrared 1	  1920x1080	@ 15Hz	   Y16
    Infrared 2	  1920x1080	@ 15Hz	   Y16
    Infrared 1	  1920x1080	@ 15Hz	   Y8
    Infrared 2	  1920x1080	@ 15Hz	   Y8
    Infrared	  1280x720	@ 30Hz	   UYVY
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared	  1280x720	@ 30Hz	   BGRA8
    Infrared	  1280x720	@ 30Hz	   RGBA8
    Infrared	  1280x720	@ 30Hz	   BGR8
    Infrared	  1280x720	@ 30Hz	   RGB8
    Infrared	  1280x720	@ 15Hz	   UYVY
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared	  1280x720	@ 15Hz	   BGRA8
    Infrared	  1280x720	@ 15Hz	   RGBA8
    Infrared	  1280x720	@ 15Hz	   BGR8
    Infrared	  1280x720	@ 15Hz	   RGB8
    Infrared	  1280x720	@ 6Hz	   UYVY
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared	  1280x720	@ 6Hz	   BGRA8
    Infrared	  1280x720	@ 6Hz	   RGBA8
    Infrared	  1280x720	@ 6Hz	   BGR8
    Infrared	  1280x720	@ 6Hz	   RGB8
    Infrared 2	  960x540	@ 25Hz	   Y16
    Infrared 1	  960x540	@ 25Hz	   Y16
    Infrared 1	  960x540	@ 15Hz	   Y16
    Infrared 2	  960x540	@ 15Hz	   Y16
    Infrared	  848x480	@ 90Hz	   UYVY
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared	  848x480	@ 90Hz	   BGRA8
    Infrared	  848x480	@ 90Hz	   RGBA8
    Infrared	  848x480	@ 90Hz	   BGR8
    Infrared	  848x480	@ 90Hz	   RGB8
    Infrared	  848x480	@ 60Hz	   UYVY
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared	  848x480	@ 60Hz	   BGRA8
    Infrared	  848x480	@ 60Hz	   RGBA8
    Infrared	  848x480	@ 60Hz	   BGR8
    Infrared	  848x480	@ 60Hz	   RGB8
    Infrared	  848x480	@ 30Hz	   UYVY
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared	  848x480	@ 30Hz	   BGRA8
    Infrared	  848x480	@ 30Hz	   RGBA8
    Infrared	  848x480	@ 30Hz	   BGR8
    Infrared	  848x480	@ 30Hz	   RGB8
    Infrared	  848x480	@ 15Hz	   UYVY
    Infrared 2	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared	  848x480	@ 15Hz	   BGRA8
    Infrared	  848x480	@ 15Hz	   RGBA8
    Infrared	  848x480	@ 15Hz	   BGR8
[... 152 lines cut ...]
    Depth	  848x480	@ 60Hz	   Z16
    Depth	  848x480	@ 30Hz	   Z16
    Depth	  848x480	@ 15Hz	   Z16
    Depth	  848x480	@ 6Hz	   Z16
    Depth	  640x480	@ 90Hz	   Z16
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  640x360	@ 90Hz	   Z16
    Depth	  640x360	@ 60Hz	   Z16
    Depth	  640x360	@ 30Hz	   Z16
    Depth	  640x360	@ 15Hz	   Z16
    Depth	  640x360	@ 6Hz	   Z16
    Depth	  480x270	@ 90Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  424x240	@ 90Hz	   Z16
    Depth	  424x240	@ 60Hz	   Z16
    Depth	  424x240	@ 30Hz	   Z16
    Depth	  424x240	@ 15Hz	   Z16
    Depth	  424x240	@ 6Hz	   Z16
```

vhanded, 2019-07-02:
@ev-mp 

The full code:
```
int width_ = 1280;
int height_ = 720;
const int rate = 30;
const rs2_format format = RS2_FORMAT_RGB8;
const rs2_stream stream = RS2_STREAM_INFRARED;
configuration_->enable_stream(stream, index, width_, height_, format, rate);
pipeline_->start(*configuration_);   // <--- It failed here. Reason: Couldn't resolve requests
```

ev-mp, 2019-07-10:
@vhanded hi, requesting IR stream to produce and RGB image requires to use IR index 0.
Please change the index and update

vhanded, 2019-07-12:
@ev-mp Thanks for your comment. It works now!

closed 2019-07-12 by vhanded