realsenseai/librealsense #4334: Intel realsense dying

BritskNguyen, 2019-07-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

BritskNguyen, 2019-07-01:
Dear everybody, I have had 3 realsense cameras bricked in a row (2 D435, 1 R200).

They are mounted on a drone, and it's usually ok for a few runs. But at a certain time, they just die, lsusb returns no device.

It may be some electrical interference here. I wonder if someone has experienced this kind of sensitivity and if you have any insights on how to overcome this? Or generally how sensitive are these sensors?

MartyG-RealSense, 2019-07-01:
Are the cameras permanently disabled (as 'bricked' implies) or do they work again after being reset, please?

I cannot think of much that would permanently destroy a RealSense camera on such a predictable schedule.  An electro magnetic pulse (EMP) can knock out electronics, but surely the rest of the drone would be affected if that were the case.  A more probable cause would be if extreme vibration was being transmitted through the drone's frame to the camera.

The article below on the website of Mouser, an Intel Authorized Distributor company, is interesting.

https://eu.mouser.com/applications/harsh-environment-vibration-protection/

If a camera is mounted on a device that is in harsh operating conditions with high vibration (whether a drone or a mobile robot on rough terrain) then some kind of vibration dampening pad on the camera's mounting is a good idea.

BritskNguyen, 2019-07-01:
They cannot be detected when I plug them on other computers. No "intel corp" device shows up when using the command lsusb.

Vibration is not the issue since we haven't flown the drone, just some ground test.

If this helps, the camera is connected to a UP board's USB. We're down to the last realsense now.

MartyG-RealSense, 2019-07-01:
I have not heard of a case like this before.  If you are plugging the camera into an original Up Board with a USB OTG micro-port then I am aware that cameras may experience difficulties operating sometimes unless a mains-powered USB hub is used, because of the camera being provided with insufficient power otherwise (the newer and more powerful Up Squared board with full USB 3.0 port does not seem to have this issue).  I have not come across a case where an Up Board could kill a camera though.

You are in a tricky situation, because if you do any kind of practical experimentation with your drone setup then you risk losing your last camera.

USB devices can heat up severely if the USB port that they are plugged into is defective.  You would probably have to run your remaining camera at least once and hold your hand on it to test whether this is the case though.  If you have another USB device that you can afford to lose (such as a USB storage stick), maybe you could test with that instead and see if it quickly gets hot while in the port.

closed 2019-07-31 by RealSenseCustomerSupport