realsenseai/librealsense #4335: cmake error (python wrapper)

ssomda21, 2019-07-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (5.11.6) | 
| Operating System & Version |   Ubuntu 16.04 LTS  | 
| Kernel Version (Linux Only)    |  (4.15.0-45)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello.

I got a problem with installation python wrapper.
When I cmake in /librealsense/wrappers/python/build
I got this error:

===========================================================================
$cmake ../ -DBUILD_PYTHON_BINDING=bool:true
------------------------------------------------------------------------------------------------------------------------
CMake Warning at CMakeLists.txt:7 (message):
  Python Bindings being built despite unset option because they are required
  for python documentation

-- pybind11 v2.2.1
CMake Error at CMakeLists.txt:16 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at CMakeLists.txt:57 (if):
  if given arguments:

    "STREQUAL" "RS2_USE_V4L2_BACKEND"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
See also "/home/smartfarm/librealsense/wrappers/python/build/CMakeFiles/CMakeOutput.log".

===========================================================================

How can I fix it? 

Thank you.

ev-mp, 2019-07-01:
@ssomda21 hello, 
Invoking CMake should always be done from [the SDK's topmost directory](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source). #4169
Please rerun and update

ssomda21, 2019-07-02:
You mean the root directory, right?
Thank you for your comment. It works well!

closed 2019-07-02 by ev-mp

danvica, 2019-08-28:
I'm using windows 10 and I've got exactly the same issue.

I've already tried to execute cmake-gui from several path including "C:\Program Files (x86)\Intel RealSense SDK 2.0" and "C:\Program Files (x86)\Intel RealSense SDK 2.0\librealsense-master".

Any suggestion ? Many thanks.