realsenseai/librealsense #4336: Reconnection Issue with T265

eMrazSVK, 2019-07-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.4.38 | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            | 2 | 
| Language                            |  C++                         | 
| Segment			|  Robot | 

### Issue Description
Hello,

I am trying to handle reconnection of T265 camera (plug/unplug USB cable of T265). When I run [rs-hello-realsense](https://github.com/IntelRealSense/librealsense/blob/master/examples/hello-realsense/rs-hello-realsense.cpp) example with D435 and disconnect D435 while the program is running, I get proprietary error `Frame didn't arrived within 15000`, which is completely okay and I am able to catch it with **try/catch** mechanism, so I can call some method which will try reconnecting the D435 in **catch** block. That works fine. Hovewer, when I do that with T265 (which should act the same, at least in my opinion based on librealsense nature) - while running [rs-pose](https://github.com/IntelRealSense/librealsense/blob/master/examples/pose/rs-pose.cpp) example for instance, I get different error and I am not able to catch it, because I get **segmentation fault**.

> 09:45:15.797 [13548] [E] Device-17A0: FW crashed - got error in stream endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
09:45:15.857 [13547] [E] Device-17A0: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
09:45:15.997 [13521] [E] Device-17A0: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
09:45:15.997 [13521] [E] Device-17A0: Entered state [ERROR_STATE]
Segmentation fault (core dumped)

After that, program terminates and I cannot do anything else about it. T265 works just fine, I am able to get streams, but I really need to implement this reconnection ability.

Could you please assist? Let me know if you need any kind of additional info from me.

Thanks!

peterhinson, 2019-07-08:
Hi @eMrazSVK, 
I had a similar issue under MacOS: after reconnecting/reinitializing the T265, the first few libusb calls in `Device::interruptEndpointThread()` would return LIBUSB_ERROR_IO, sending the driver into ERROR_STATE (the handling of which seems to bork the device). 

As a hack, simply ignoring LIBUSB_ERROR_IO appears to allow reconnection. In my case the error is issued a few times and then communication resumes. See this [commit](https://github.com/peterhinson/librealsense/commit/7cca0eb00c7298ee9108b5862656b5def90f53a8). May need to be applied to `Device::streamEndpointThread()` as well.

eMrazSVK, 2019-07-08:
@peterhinson Thanks for answer, but I still get `Segmentation fault (core dumped)`. I applied your "patch" to both `Device::interruptEndpointThread()` and `Device::streamEndpointThread()`. Still no success. I wonder how is this possible. In the Intel's realsense-ros wrapper it is somehow handled and the process doesn't terminate.

peterhinson, 2019-07-09:
Shoot! You could try enabling all logging levels in libtm to narrow it down. Is the program entering ERROR_STATE prior to the seg fault? I suspect something is left dangling/stale in that code path- in my case, the driver returns bogus sensor readings and fails to resync the clock even after exiting ERROR_STATE.

It's interesting that the ROS wrapper doesn't exhibit the issue. By the looks of the [ROS RealsenseNodeFactory class](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/realsense_node_factory.cpp), they perform quite a bit more error checking than the SDK examples...

eMrazSVK, 2019-07-09:
@peterhinson Thanks, will try that libtm logging to trace the error. And based on the output in terminal - yes ERROR state is entered prior to seg fault, but that might be only the output, I do not know about exact behaviour, must check that.

Btw I have a feeling, that in ros-wrapper, they just shutdown the node and create new one before the segmentation fault happens. In my opinion, it should somehow work out of the box - even with the simplest example I do not get seg fault when using D435. Seg fault is something, that should never happen.

RealSenseCustomerSupport, 2019-07-10:
----------------------------------------------

Hi @eMrazSVK,

1. which librealsense version do you use now?
2. Per below message, it should be due to the USB connection. You plan to design a mechanism to handle the disconnect event?
09:45:15.797 [13548] [E] Device-17A0: FW crashed - got error in stream endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0

3. For libtm host log, you can check the code in manager.cpp in libtm library. Change the log level from Error to Debug as follows.

    Manager::Manager(Listener* lis, void* param) : mDispatcher(new Dispatcher()), mListener(nullptr), mContext(nullptr), mFwFileName(""), mLibUsbDeviceToTrackingDeviceMap(), mEvent(), mCompleteQMutex(), mCompleteQ(), mTrackingDeviceInfoMap()
    {

        setHostLogControl({ LogVerbosityLevel::Debug, LogOutputMode::LogOutputModeScreen, true });

eMrazSVK, 2019-07-10:
Hi @RealSenseCustomerSupport 

1. I use 2.23, but I triead 2.24 and no success

2. I was planning it, but for my project, it is not so important right now, so I dropped this interest for now

3. Thanks, I can try this at least

Btw this happens also on standard Ubuntu laptop as well. 

Any additional information I can provide:

- I get segmentation fault when I unplug T265 while running example `rs-pose`
- When I modify `rs-pose` example to use `poll_for_frames()` function instead of `wait_for_frames()` I do not get segmentation fault. So the segmentation fault happens because the code tries to access something in a pipeline what is not available anymore due to disconnection. Anyway, the `LIBUSB_ERROR_IO` persists.

closed 2019-07-22 by eMrazSVK

seratg, 2019-07-22:
I guess we have a similar issues which I just reported: https://github.com/IntelRealSense/librealsense/issues/4475

Even when using poll_for_frames(), unplugging the camera causes the application to crash upon releasing the RealSense resources. The stack trace is shown in the issue.

Have you found any solutions ? 

Thanks

zainmehdi, 2019-10-25:
Any solution for this ??

shreyas2311, 2020-01-15:
Is this issue solved? Why is it closed?

eMrazSVK, 2020-01-15:
@shreyas2311 I am just a user, I do not have capacity to design connect/disconnect mechanism. It is closed, because I lost interest in solving this, since it is not important in my project anymore. BUT if you want, I could reopen it for you.

shreyas2311, 2020-01-16:
Hi @eMrazSVK Thanks for the reply. I have the same issue, however, it works fine when I disable streaming the fisheye camera views. I think, one USB 3.0  port on the TX2 isn't able to handle the D435 and the T265 together. I guess you could leave the issue opened which would indicate that the issue persists. Thanks a lot!