realsenseai/librealsense #4337: D415 depth image resolution

sailfish009, 2019-07-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.23.0 }                          | 
| Language                               |  {C                         | 
| Segment			        |  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hi, I am trying to save 1270x720 depth image to file(*.PNG).
but when i saved the image using intel realsense-viewer, 
the image file resolution was 640x360.

i want to view/save depth image resolution as 1270x720.

ev-mp, 2019-07-01:
@sailfish009 hello, 
This is often a result of [decimation post-processing filter](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter) being turned on in the Viewer.  #3392, #3989

sailfish009, 2019-07-01:
@ev-mp thanks! it works fine for me!

closed 2019-07-01 by sailfish009