realsenseai/librealsense #4341: "Failed to identify Internet connection" but connection works

aPonza, 2019-07-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.11.01.100 prior to failure| 
| Operating System & Version | Ubuntu 16.04  | 
| Kernel Version (Linux Only) | 4.8.15-rt10 PREEMPT RT | 
| SDK Version                            |  2.22.0 prior to failure                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                         | 

### Issue Description
After updating to master, I tried to rebuild from scratch and I'm seeing this:

```
<path>/librealsense/build$ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo 
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMake/connectivity_check.cmake:3 (file):
  file DOWNLOAD cannot open file for write.
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Failed to identify Internet connection
-- Info: REALSENSE_VERSION_STRING=2.24.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- No internet connection, disabling BUILD_WITH_TM2
-- using RS2_USE_V4L2_BACKEND
-- Found usb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- GLFW 3.3 not found; using internal version
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so   
-- No internet connection, disabling IMPORT_UVC_FW
-- Configuring incomplete, errors occurred!
See also "<path>/librealsense/build/CMakeFiles/CMakeOutput.log".
```

The internet connection is definitely working (I'm posting this issue) and I don't get why internet absence would want to kill the build since the firmware update could be done at a separate moment.

Commenting [this line](https://github.com/IntelRealSense/librealsense/blob/6a7917c2bdcc3cb5c93b161c876c632d73634866/CMakeLists.txt#L5) results in 

```
<path>/librealsense/common/fw-update-helper.cpp:15:10: fatal error: common/fw/D4XX_FW_Image.h: No such file or directory
 #include "common/fw/D4XX_FW_Image.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

which is included due to `#ifdef INTERNAL_FW` which I can't find being defined anywhere, and I also can't find the to-be-included file: the closest match is `D4XX_FW_Image.rc` which also doesn't exist but is included in [here](https://github.com/IntelRealSense/librealsense/blob/f112ed78e3917df79047254f864157195d4488b0/common/fw/uvc_fw.rc#L1).

ev-mp, 2019-07-02:
@aPonza hello,
Iasit possible that you're using a corporate network ? In that case you'll probably need to configure a proxy to bypass the firewalls. Alternatively - obtain an uncensored internet access (use Phone's hotspot), rerun the cmake and update .

aPonza, 2019-07-03:
That doesn't seem to be it. I'm already going out skipping the company firewall, but I still tried: while connected with my phone, I uncommented the `include(CMake/connectivity_check.cmake)` line and ran this:
```
<path>/librealsense/test_build$ cd .. && sudo rm -rf test_build && mkdir test_build && cd test_build
[sudo] password for <user>: 
<user>@<loc>:<path>/librealsense/test_build$ cmake .. -DCMAKE_BUILD_TYPE=Debug
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
CMake Error at CMake/connectivity_check.cmake:4 (file):
  file DOWNLOAD cannot open file for write.
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Failed to identify Internet connection
-- Info: REALSENSE_VERSION_STRING=2.24.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- No internet connection, disabling BUILD_WITH_TM2
-- using RS2_USE_V4L2_BACKEND
-- Found usb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so   
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- No internet connection, disabling IMPORT_UVC_FW
-- Configuring incomplete, errors occurred!
See also "<path>/librealsense/test_build/CMakeFiles/CMakeOutput.log".
```

...so the same output as before. I also tried:
```
$ wget "http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/connectivity_check"
--2019-07-03 10:11:14--  http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/connectivity_check
Resolving realsense-hw-public.s3-eu-west-1.amazonaws.com (realsense-hw-public.s3-eu-west-1.amazonaws.com)... 52.218.96.139
Connecting to realsense-hw-public.s3-eu-west-1.amazonaws.com (realsense-hw-public.s3-eu-west-1.amazonaws.com)|52.218.96.139|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 0 [text/plain]
Saving to: ‘connectivity_check’

connectivity_check                                     [ <=>                                                                                                             ]       0  --.-KB/s    in 0s      

2019-07-03 10:11:15 (0,00 B/s) - ‘connectivity_check’ saved [0/0]
```
and with both connections the output is the same, so that's not it as well. What else can I try?

ev-mp, 2019-07-04:
@aPonza , thanks for the logs.
It is not clear where the permission issue is coming from and it could be that the build directory is somehow corrupted. I'd recommend to make a clean cmake build and then post the full cmake log:
From Librealsense root dir :
`cd build`
`sudo make uninstall && make clean && sudo rm -rf * && cmake .. -DBUILD_EXAMPLES && make -j$(nproc) && sudo make install`

aPonza, 2019-07-04:
Stops at the configuration phase with the same output as before: the "See also" log file is [CMakeOutput.log](https://github.com/IntelRealSense/librealsense/files/3359672/CMakeOutput.log)

hpf9017, 2019-07-05:
I met the same problem before.
the reason is that you are lack of target-0.0.18.*.mvcmd file.
you need to wait for "Build with TM2" to download the file compeletely when you :
cmake ../ in the build directory.
[image]

aPonza, 2019-07-05:
Not sure I understand:
- I set the waiting time to 60s in [here](https://github.com/IntelRealSense/librealsense/blob/6a7917c2bdcc3cb5c93b161c876c632d73634866/CMake/connectivity_check.cmake#L3) but nothing changed;
- Per normal build configuration, this line happens `-- No internet connection, disabling BUILD_WITH_TM2` before I reach the bit you're highlighting in the image, and this leads me to believe your error is kind of different from mine.

hpf9017, 2019-07-08:
when I run the cmake ..
it stay at “Building with TM2” for a long time.
I think it took more than 10 minutes with my internet speed.
actually, I went to do others things,
then come back it finally finished.

guikunchen, 2019-07-10:
@hpf9017 hi, I meet the same problem looks like the image you had post. "Building with TM2" takes 30 minutes or more, when it finished, the problem emerge. Could you give me some help? Thank you for your time.

hpf9017, 2019-07-11:
@guikunchen this is what happened when I took this picture.
then I changed the internet with a VPN and then it successed.

aPonza, 2019-07-11:
@guikunchen @hpf9017 your problem is definitely different from mine, I don't get to "Building with TM2" at all, and in fact it's set to **not** build with TM2, currently. I'd suggest opening a separate issue if yours is still not resolved.

hpf9017, 2019-07-11:
@aPonza you can try to use cmake .. -DCMAKE_BUILD_TYPE=Release 
according to [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md](url) 
in 'Building librealsense2 SDK'
I use your command 'cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo', and get the same
fault with yours.

aPonza, 2019-07-11:
I don't understand how this makes sense, but it is correct. In relation to [this comment](https://github.com/IntelRealSense/librealsense/issues/4341#issuecomment-508347900), I'm also not sure, since it should have removed the cache with the listed command? Thank you @hpf9017!

hpf9017, 2019-07-12:
@aPonza not at all. I don't know either. Need other people to explain.

ev-mp, 2019-09-19:
@aPonza , @hpf9017 - did you manage to resolve the connectivity issues?

aPonza, 2019-09-19:
It works now, not sure what changed though, but configuring with      `-DCMAKE_BUILD_TYPE=RelWithDebInfo`, building and installing works: I'm closing the issue

closed 2019-09-19 by aPonza