realsenseai/librealsense #4343: DKMS not correctly patching, Ubuntu 18.04

atyshka, 2019-07-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 4.18.0-25-generic| 
| Platform                                 | PC |
| SDK Version                            |  2.24.0 | 
| Language                            | NA | 
| Segment			| Robot | 

I am on Ubuntu 18.04 and was having trouble installing the DKMS. Today I saw the 2.24.0 release included support for kernel 4.18, so I was excited to try it out. For good measure, I apt-removed the whole dkms package and fresh installed it. Here's my log:

```
Unpacking librealsense2-dkms (1.3.5-0ubuntu1) ...
Setting up librealsense2-dkms (1.3.5-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.5 DKMS files...
Building for 4.18.0-25-generic
Building initial module for 4.18.0-25-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/librealsense2-dkms.0.crash'
Error! Bad return status for module build on kernel: 4.18.0-25-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.5: added
nvidia, 418.67, 4.18.0-24-generic, x86_64: installed
nvidia, 418.67, 4.18.0-25-generic, x86_64: installed
mod: videodev 
mod: uvcvideo                    version: 1.1.1
mod: hid_sensor_gyro_3d 
mod: hid_sensor_accel_3d 
```
So I checked out the crash report:

cat /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log 
```
DKMS make.log for librealsense2-dkms-1.3.5 for kernel 4.18.0-25-generic (x86_64)
Mon Jul  1 10:50:04 EDT 2019
make: Entering directory '/usr/src/linux-headers-4.18.0-25-generic'
arch/x86/Makefile:155: CONFIG_X86_X32 enabled but no binutils support
arch/x86/Makefile:244: *** You are building kernel with non-retpoline compiler, please update your compiler..  Stop.
make: Leaving directory '/usr/src/linux-headers-4.18.0-25-generic
```
What is causing this error and how can I fix it?

jpapon, 2019-07-03:
Exact same problem here on a fresh install of 18.04.02. Anyone find a workaround?

ev-mp, 2019-07-04:
@jpapon , we are struggling to upload to the server, in the meanwhile please install it manually from the attached file:
`mv librealsense2-dkms_1.3.6-0ubuntu0_all.txt librealsense2-dkms_1.3.6-0ubuntu0_all.deb`
`sudo dpkg -i librealsense2-dkms_1.3.6-0ubuntu0_all.deb`
[librealsense2-dkms_1.3.6-0ubuntu0_all.txt](https://github.com/IntelRealSense/librealsense/files/3358091/librealsense2-dkms_1.3.6-0ubuntu0_all.txt)

ev-mp, 2019-07-04:
@atyshka ,@jpapon,
The package for Ubuntu 18.4.2 with kernel 4.18 is now available. To upgrade the current version to the new 1.3.6 use -
`sudo apt update && sudo apt upgrade`

jpapon, 2019-07-04:
Great, thank you.
I can't test at the moment, but I had no problem with master compiled from source, so I'm assuming the package works too.

ev-mp, 2019-07-10:
@atyshka ,@jpapon, did you manage to deploy and test the package ?

jpapon, 2019-07-10:
@ev-mp Yes, it worked fine for me on Kubuntu 18.04.02

ev-mp, 2019-07-11:
Thank you for feedback!

closed 2019-07-11 by ev-mp